Ralim/IronOS #814: Iron never negotiates 12V over PD (TS80P)

TheHamster95, 2021-01-19:
This is a Bug

* **I have**
  - [*] Searched previous issues 
  - [*] This is in _this_ firmware, not vendor "offical" firmware
  - [*] This is not a bug in the vendor bootloader (aka DFU)
  - [*] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?** *
Plug iron into USB PD charger/ powerbank that supports 12V. Debug Menu shows it's running PD, but voltage never goes above 9.1V

* **What is the expected behavior?**
Iron should negotiate 12V PD.

***Steps to reproduce the bug:***

- Plug iron into PD source
- Check voltage

  - Firmware Version: 2.13.CFB9239
  - PCB Version: 4
  - Power Supply (Voltage and Current Rating) : 12V/1.5A

alvinhochun, 2021-01-19:
The TS80P firmware has it [hard-coded](https://github.com/Ralim/IronOS/blob/f3488d7ee354cf3a5c617282a8bd4b09c3edd9ea/source/Core/BSP/Miniware/BSP_PD.c#L16) that 12V will require 2.6A capability, because the tip heater is a resistive load and *will* sink this amount of current at 12V. As your PSU can only supply 1.5A at 12V it won't get used.

TheHamster95, 2021-01-19:
Ah, didn't see that anywhere. Will close then.

closed 2021-01-19 by TheHamster95