Ralim/IronOS #815: TS100 from minihere reading way off

alemoh1234, 2021-01-19:
_This is a [Question]_

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
room temperature is reading about 65-72 celsius, even with a different tip (just bought my setup and as soon as it arrived I changed the firmware, readings seem stable and it works, I have a second tip, and upon changing it the temperature reading remains wrong

* **What is the expected behavior?**
correct temperature reading

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.14.2425902
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : Came with the soldering iron, 24V 3A

Is this an issue with my device? should I try to return it?

Vinigas, 2021-01-19:
Have you tried calibrating it yet ?

[IronOS->Documentation->Menu.md](https://github.com/Ralim/IronOS/blob/master/Documentation/Menu.md)

alemoh1234, 2021-01-19:
> o Have you tried calibrating it yet ?
I will give it a try, but I have no high precision thermometers, so that`s why I had not considered that option, also because the measurements were so wrong.

> o Is tip from official providers ?
Yes, also from miniwere.

> o Which tip model are you using ?
TS-I and TS-D24.

> o Does this happens with all kinds of tips ?
Well, as stated above, with both tips I have the same thing happens, so yes.

> o Could you output here everything what is written in debug-menu ?

Will do, as soon as I get to id I will edit this and post it here.

Ralim, 2021-01-19:
Hello,

Yep as stated above, please provide the output of the debug menu.
This _sounds like_ either something is wrong with the handle temp sensor reading or Miniware have randomly changed a part on us again without any warning or documentation.

alemoh1234, 2021-01-19:
Thanks for answering Ralim, I have to say your firmware is awesome, I spent some good 15 minutes just tinkering with the options.
As soon as I can I will be posting the debug results

alemoh1234, 2021-01-19:
HW G 182
HW M 189
HW P 96
TIME 950
MOVE 646
RTip 800
CTip 60
CHan 610
Vin 0.0
PCB 2
PWR 0
Max 445

these were taken while powered by USB on a slightly colder room.

Ralim, 2021-01-19:
Aha, yes that confirms that the issue is the handle (cold junction) temperature sensor is reporting 61C.
Which is _very_ out of whack.

Is it possible for you to chill the unit down a bit (say leave in a fridge in a sealed container for an hour) and take these ones again:
- RTip 
- CTip 
- CHan 

Will try and use two approx temps to see if its a faulty part or if its something they have changed :)

alemoh1234, 2021-01-19:
Copy that, do you recommend I give you a temperature measurement of the fridge? because unfortunately I have not got a good thermometer that can read negatives, I will just be able to approximate it by what the freezer manufacturer says.

Ralim, 2021-01-19:
Estimated is fine, only looking for a rough delta to get a feel for what kind of sensor it is.

alemoh1234, 2021-01-19:
According to the freezer manufacturer, -18°C give or take 3°C, the TS100 has been sitting in the freezer for some 30 minutes now, I will give it 30 more and report back.

alemoh1234, 2021-01-20:
Well, something does not like to be cold, once plugged it the oled has what seems to be a random noise patter, and does not work, I just hope it works again once it warms up

alemoh1234, 2021-01-20:
RTP 950
CTIP 53
CHAN 535

I estimate it was at about 15C once it came back to life.

alemoh1234, 2021-01-20:
I have put it back in the freezer but powered on with a power bank, lets see if this time around I get some better measurements.

alemoh1234, 2021-01-20:
got theese

Rtip 675
Ctip 43
Chan 435

not sure if I can get much colder with it still working

Ralim, 2021-01-20:
Thats okay, even we with 15C we are seeing a temperature change going on there, but its no where near as sensitive as it should be.

If you flash the official firmware does it read _correctly_?

alemoh1234, 2021-01-20:
Well, correctely no, but closer to what the correct should be, yes, but not by much as stated before.

> edit: upon flashing the original firmware version 2017.11.24 it reads 48C room temp (edited because it had stayed for some minutes in front of the A/C when flashing, new number is now on the same conditions as before).

at the time of this edit with your firmware it was measuring about 60C

alemoh1234, 2021-01-20:
got this while starting to artifact the oled, I have now taken it out of the freezer, for sure it was near zero C

Rtip 665
Chan 358
Ctip 36

Ralim, 2021-01-20:
If it's not working on the miniware firmware as well I would contact them for a replacement as it seems more likely that the unit has a damaged temperature sensor than anything else.

If they are unhelpful it is possible to replace it yourself but I feel they should replace this unit :/

alemoh1234, 2021-01-20:
Thank you very much, I hope they are reasonable and help me out.

once again, grate firmware, you are a blessing to the open source community.

closed 2021-01-20 by alemoh1234