Ralim/IronOS #817: Publish firmware on Linux Vendor Firmware Service

t-8ch, 2021-01-20:
The Linux Foundation operates a online repository of device firmware (the Linux Vendor Firmware Service, LVFS) at https://fwupd.org/ .
This repository is currently consumed by the fwupd tool on Linux which provides a easy one-click update-experience to update device firmware.

It would be great if IronOS could be uploaded there and be at least published as the default firmware for the Pinecil.
As fwupd already supports the USB DFU protocol that part should be reasonably straight forward.
Some context: https://twitter.com/dal000/status/1344630316015755265

The following open points I see:
1) IronOS has one firmware build for each language. This does not seem to be supported in LVFS/fwupd currently, so it would need development there.
2) fwupd uses its own DFU implementation, so it will also need a custom quirk implementation to handle the wrong blocksize reported by the Pinecil DFU implementation (as dfu-util is GPL2 and fwupd is LGPL-2.1+ I don't think their implementation can be used as-is)
3) IronOS can be used as unofficial firmware for other soldering irons, is this something that can/should be supported by LVFS?
4) LVFS does not have a specific icon for soldering icons!

@Ralim  is this something you would want to have?
@hughsie As maintainer of fuwpd and LVFS what is your take on this?

Relevant issue for Pine64s official firmware updater: https://github.com/pine64/pinecil-firmware-updater/issues/1

hughsie, 2021-01-20:
> IronOS has one firmware build for each language

So there are a few different ways for USB devices you can get a "custom" GUID. The easiest is to have a different USB PID for each language; that makes everything happen correctly with no code changes, although kinda sucks from a build-time complexity problem. I have a USB VID you can use if you want so certainly don't spend the $2000 to just get a new VID assigned -- and it has 0xfff2 PIDs remaining :)

The other way is to somehow tell fwupd what language the device is in; maybe using the extended descriptor thing from Microsoft or just something custom we have to add code for. You could even stick it in the serial number string index or something. Basically; if the device can export the data we can add code to fwupd to consume it.

> so it will also need a custom quirk implementation to handle the wrong blocksize

So `desc.wTransferSize` is wrong? I can easily add a quirk to set that -- what's the VID/PID of the bootloader and what should the wTransferSize be in reality?

> IronOS can be used as unofficial firmware for other soldering irons

If it helps, the LVFS already supports "branches" -- e.g. on the BC57XX network cards there's an unofficial open source firmware and the nonfree broadcom one and you can "fwupdmgr swithc-branch" between them.

> LVFS does not have a specific icon for soldering icons

Ha! Got any ideas? it needs to be freely re-distributable and in the hicolor theme.

More seriously, I think the bigger problem is that the firmware doesn't seem to have a runtime; that is when you plug in the soldering iron you don't see a USB device being added, you only see one when the device is in bootloader mode. This means that fwupd can't get the current firmware version without the user manually going into bootloader mode first, which kinda sucks as:

 * You don't get notified about new firmware
 * The bootloader VID/PID can't be modified so you don't know what hardware you have attached

@Ralim seems to be open to the idea of a runtime interface for monitoring, and it would be trivial to implement a DFU runtime with all the required bits so that fwupd can enumerate the hardware to get the language and current firmware version and also automatically push it into bootloader mode. I think that's the prerequisite before we start talking about icons :)

t-8ch, 2021-01-20:
> > so it will also need a custom quirk implementation to handle the wrong blocksize
> 
> So `desc.wTransferSize` is wrong? I can easily add a quirk to set that -- what's the VID/PID of the bootloader and what should the wTransferSize be in reality?

I don't think it's `wTransferSize` that is wrong.
It's more the sector size and count that is misreported.
This is the GPL2 patch to dfu-util: https://sourceforge.net/p/dfu-util/dfu-util/ci/f2b7d4b1113ef6c3ada31a0654c9aefebcdb1de5/
The bootloader has idVendor=28e9, idProduct=0189 .

hughsie, 2021-01-20:
Ahh, it's a strange beast indeed. I've started the process with https://github.com/hughsie/libgusb/pull/43

hughsie, 2021-01-20:
100% untested incomplete WIP branch: wip/hughsie/GD32VF103 -- I've ordered a Sipeed Longan Nano dev board and can finish off the branch when it arrives.

hughsie, 2021-01-27:
Seems to work for me: https://github.com/fwupd/fwupd/pull/2815 -- there's the slight issue that fwupd doesn't know how to flash an Intel Hex file to the DFU device, but I can work on that too.

Ralim, 2021-01-28:
@hughsie We do also pack the `bin` file in all releases so `hex` support shouldnt be required for now.

hughsie, 2021-01-28:
Ahh, I missed that. The bin file might need fwupd work as I'm assuming the base offset is 0 rather than 0x8000000 or whatever the chip has. Got a link handy? Also, did you ponder a runtime USB interface?

Ralim, 2021-01-28:
Ah yes the bin files are setup to be flashed at the start of flash which is indeed 0x8000000.

As an example, any pinecil.zip from the actions tab will work or this release https://github.com/Ralim/IronOS/releases/download/v2.14.1/Pinecil.zip

Heh, yeah have been thinking about adding a usb mode. At the least even just all of the descriptors would be useful for this.

Just haven't sat down to code it yet sadly 😅

dal00, 2021-02-17:
I see support is now merged in the latest fwupd-release now! That is great! Does that mean it is nearing completion? Or is there still outstanding work to be done on the Pinecil site of things?
I mean the usb mode might not be required for initial support? That would only make it a nicer user experience?

alvinhochun, 2021-02-17:
> I mean the usb mode might not be required for initial support? That would only make it a nicer user experience?

Without a USB mode, how would fwupd identify the Pinecil? I don't think we can just use the GD32 DFU VID/PID pair. The Pinecil is not the only thing that uses the GD32VF103 MCU and we wouldn't want to accidentally flash IronOS to some other devices.

hughsie, 2021-02-17:
> how would fwupd identify the Pinecil

We need a USB runtime interface, even if it's just the interface with some revision number and a VID PID. fwupd is smart enough to know that _Pinecil_ -> _generic DFU bootloader_ = "_Pinecil in bootloader mode_", although without the runtime with unique VIDPID you're correct in that we can't do anything correctly.

hughsie, 2021-02-26:
> Just haven't sat down to code it yet sadly

I tried to flash the UK Pinecil firmware to the GD32VF103 I have in the Sipeed Longan Nano but that seems to have `0x08000000/128*001Kg` i.e. 32kb of flash and only ~60% of the flash fits. Does the Pinecil have more flash than this or have I got my sums wrong? The Nano has a ChipID of `0603` if that helps.

t-8ch, 2021-02-26:
@hughsie
The Pinecil should have 128KB of flash ( https://wiki.pine64.org/index.php/Pinecil ).
If you tell me how to retrieve the ChipID I can check it for the Pinecil.

hughsie, 2021-02-26:
> If you tell me how to retrieve the ChipID I can check it for the Pinecil

Install libgusb fwupd from git master (or the F33 packages from https://people.freedesktop.org/~hughsient/fedora/ -- both fwupd and libgusb) and then `sudo fwupdtool get-devices --plugins dfu --verbose` with the Pinecil in bootloader mode. If that all sounds like too much hard work I'm pretty sure the pinecil has about twice the flash I have on my devboard.

t-8ch, 2021-02-27:
ChipID seems for the Pinecil seems to be `0603`, too.

<details>
  <summary>Full log</summary>

```
23:56:55:0729 FuEngine             using plugins: dfu
23:56:55:0758 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
23:56:55:0758 FuPluginDfu          using DFU transfer size 0x0040 bytes
23:56:55:0758 FuPluginDfu          unable to guess initial device state from interface 0
23:56:55:0758 FuPluginDfu          STM-DFU support
23:56:55:0758 FuPluginDfu          using DFU transfer size 0x0800 bytes
23:56:55:0758 FuPluginDfu          unable to guess initial device state from interface 0
23:56:55:0758 FuDevice             using 23ec719b6aabc2d2dac5176c232f0da7a21881b0 for usb:01:00:04
23:56:55:0868 FuPluginDfu          chip ID set to: 0603
23:56:55:0868 FuPluginDfu          using GD32 sector size of 0x8000
23:56:55:0869 FuPluginDfu          using GD32 sector size of 0x8000
23:56:55:0870 FuPlugin             emit added from dfu: 23ec719b6aabc2d2dac5176c232f0da7a21881b0
23:56:55:0871 FuDeviceList         ::added 23ec719b6aabc2d2dac5176c232f0da7a21881b0
23:56:55:0871 FuMain               ADDED:
FuDfuDevice:
GD32 0x418 DFU Bootloade?
  DeviceId:             23ec719b6aabc2d2dac5176c232f0da7a21881b0
  Guid:                 f3005d4e-2e30-503a-bc93-fcb79ce65f34 <- USB\VID_28E9&PID_0189&REV_1000
  Guid:                 f40ef7cf-e4f6-5708-b274-13edfb04646e <- USB\VID_28E9&PID_0189
  Serial:               3TBJ
  Plugin:               dfu
  Protocol:             com.st.dfuse
  Flags:                updatable|registered|add-counterpart-guids
  Vendor:               GDMicroelectronics
  VendorId:             USB:0x28E9
  Version:              10.0
  VersionFormat:        bcd
  Icon:                 drive-harddisk-usb
  Created:              2021-02-26
  PhysicalId:           usb:01:00:04
  CustomFlags:          gd32
  PossiblePlugin:       dfu
  InternalFlags:        replug-match-guid
  State:                appIDLE
  Status:               OK
  DoneUploadOrDownload: false
  ClaimedInterface:     false
  ChipId:               0603
  Version:              0x11a
  ForceVersion:         0x0
  RuntimePid:           0x189
  RuntimeVid:           0x28e9
  RuntimeRelease:       0x1000
  TransferSize:         0x800
  IfaceNumber:          0x0
  DnloadTimeout:        0x0
  TimeoutMs:            0x5dc
    AltSetting:         0x0
    AltIdx:             0x4
    AltName:            @Internal Flash  /0x08000000/512*002Kg
      Idx00:            Zone:0, Sec#:0, Addr:0x08000000, Size:0x8000, Caps:0x7 [REW]
    AltSetting:         0x1
    AltIdx:             0x5
    AltName:            @Option Bytes  /0x1FFFF800/01*016 g
      Idx00:            Zone:0, Sec#:0, Addr:0x08000000, Size:0x8000, Caps:0x7 [REW]

23:56:55:0901 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
23:56:55:0903 FuMain               GD32 0x418 DFU Bootloade?
  DeviceId:             23ec719b6aabc2d2dac5176c232f0da7a21881b0
  Guid:                 f3005d4e-2e30-503a-bc93-fcb79ce65f34 <- USB\VID_28E9&PID_0189&REV_1000
  Guid:                 f40ef7cf-e4f6-5708-b274-13edfb04646e <- USB\VID_28E9&PID_0189
  Serial:               3TBJ
  Plugin:               dfu
  Protocol:             com.st.dfuse
  Flags:                updatable|registered|add-counterpart-guids
  Vendor:               GDMicroelectronics
  VendorId:             USB:0x28E9
  Version:              10.0
  VersionFormat:        bcd
  Icon:                 drive-harddisk-usb
  Created:              2021-02-26
```
</details>

hughsie, 2021-02-27:
@t-8ch does https://github.com/fwupd/fwupd/pull/2952 fix things? I'm away from my lab today so I can actually test until Monday.

Ralim, 2021-02-27:
Ah you are correct in the different flash sizes; though the fitness can be flashed to either as it fits; just that I'm not certain what would happen when it tries to save settings as these are located at the end of flash

t-8ch, 2021-02-27:
@hughsie Now it looks different, but still 0603.
<details>
  <summary>Full log</summary>

```
12:34:16:0270 FuEngine             Emitting PropertyChanged('Status'='loading')
Loading…                 [-                                      ]12:34:16:0270 FuEngine             using plugins: dfu
12:34:16:0347 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
12:34:16:0347 FuPluginDfu          using DFU transfer size 0x0040 bytes
12:34:16:0347 FuPluginDfu          quirking device into DFU mode
12:34:16:0347 FuPluginDfu          STM-DFU support
12:34:16:0347 FuPluginDfu          using DFU transfer size 0x0800 bytes
12:34:16:0347 FuPluginDfu          quirking device into DFU mode
12:34:16:0347 FuDevice             using 23ec719b6aabc2d2dac5176c232f0da7a21881b0 for usb:01:00:04
12:34:16:0458 FuPluginDfu          chip ID set to: 0603
12:34:16:0459 FuPlugin             emit added from dfu: 23ec719b6aabc2d2dac5176c232f0da7a21881b0
12:34:16:0460 FuDeviceList         ::added 23ec719b6aabc2d2dac5176c232f0da7a21881b0
12:34:16:0461 FuMain               ADDED:
FuDfuDevice:
GD32 0x418 DFU Bootloade?
  DeviceId:             23ec719b6aabc2d2dac5176c232f0da7a21881b0
  Guid:                 f3005d4e-2e30-503a-bc93-fcb79ce65f34 <- USB\VID_28E9&PID_0189&REV_1000
  Guid:                 f40ef7cf-e4f6-5708-b274-13edfb04646e <- USB\VID_28E9&PID_0189
  Serial:               3TBJ
  Plugin:               dfu
  Protocol:             com.st.dfuse
  Flags:                updatable|registered|is-bootloader|add-counterpart-guids
  Vendor:               GDMicroelectronics
  VendorId:             USB:0x28E9
  Version:              10.0
  VersionFormat:        bcd
  Icon:                 drive-harddisk-usb
  Created:              2021-02-27
  PhysicalId:           usb:01:00:04
  CustomFlags:          gd32,force-dfu-mode
  PossiblePlugin:       dfu
  InternalFlags:        replug-match-guid
  State:                dfuIDLE
  Status:               OK
  DoneUploadOrDownload: false
  ClaimedInterface:     false
  ChipId:               0603
  Version:              0x11a
  ForceVersion:         0x0
  RuntimePid:           0xffff
  RuntimeVid:           0xffff
  RuntimeRelease:       0xffff
  TransferSize:         0x800
  IfaceNumber:          0x0
  DnloadTimeout:        0x0
  TimeoutMs:            0x5dc
    AltSetting:         0x0
    AltIdx:             0x4
    AltName:            @Internal Flash  /0x8000000/128*1Kg
    AltNameForDisplay:  Internal Flash
      Idx00:            Zone:0, Sec#:0, Addr:0x08000000, Size:0x0400, Caps:0x7 [REW]
      Idx01:            Zone:0, Sec#:0, Addr:0x08000400, Size:0x0400, Caps:0x7 [REW]
      Idx02:            Zone:0, Sec#:0, Addr:0x08000800, Size:0x0400, Caps:0x7 [REW]
      Idx03:            Zone:0, Sec#:0, Addr:0x08000c00, Size:0x0400, Caps:0x7 [REW]
      Idx04:            Zone:0, Sec#:0, Addr:0x08001000, Size:0x0400, Caps:0x7 [REW]
      Idx05:            Zone:0, Sec#:0, Addr:0x08001400, Size:0x0400, Caps:0x7 [REW]
      Idx06:            Zone:0, Sec#:0, Addr:0x08001800, Size:0x0400, Caps:0x7 [REW]
      Idx07:            Zone:0, Sec#:0, Addr:0x08001c00, Size:0x0400, Caps:0x7 [REW]
      Idx08:            Zone:0, Sec#:0, Addr:0x08002000, Size:0x0400, Caps:0x7 [REW]
      Idx09:            Zone:0, Sec#:0, Addr:0x08002400, Size:0x0400, Caps:0x7 [REW]
      Idx0a:            Zone:0, Sec#:0, Addr:0x08002800, Size:0x0400, Caps:0x7 [REW]
      Idx0b:            Zone:0, Sec#:0, Addr:0x08002c00, Size:0x0400, Caps:0x7 [REW]
      Idx0c:            Zone:0, Sec#:0, Addr:0x08003000, Size:0x0400, Caps:0x7 [REW]
      Idx0d:            Zone:0, Sec#:0, Addr:0x08003400, Size:0x0400, Caps:0x7 [REW]
      Idx0e:            Zone:0, Sec#:0, Addr:0x08003800, Size:0x0400, Caps:0x7 [REW]
[... 114 lines cut ...]
    AltIdx:             0x5
    AltName:            @Option Bytes  /0x1FFFF800/01*016 g
    AltNameForDisplay:  Option Bytes
      Idx00:            Zone:0, Sec#:0, Addr:0x1ffff800, Size:0x0010, Caps:0x7 [REW]

12:34:16:0491 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
12:34:16:0493 FuMain               GD32 0x418 DFU Bootloade?
  DeviceId:             23ec719b6aabc2d2dac5176c232f0da7a21881b0
  Guid:                 f3005d4e-2e30-503a-bc93-fcb79ce65f34 <- USB\VID_28E9&PID_0189&REV_1000
  Guid:                 f40ef7cf-e4f6-5708-b274-13edfb04646e <- USB\VID_28E9&PID_0189
  Serial:               3TBJ
  Plugin:               dfu
  Protocol:             com.st.dfuse
  Flags:                updatable|registered|is-bootloader|add-counterpart-guids
  Vendor:               GDMicroelectronics
  VendorId:             USB:0x28E9
  Version:              10.0
  VersionFormat:        bcd
  Icon:                 drive-harddisk-usb
  Created:              2021-02-27

```
</details>

hughsie, 2021-03-01:
> Now it looks different, but still 0603

Ya, as it should; I misunderstood the original quirk purpose. I've fixed up the branch and just flashed successfully Pinecil_UK.bin to the development board. It did not, _however_, magically turn into a soldering iron.

Once @Ralim adds a DFU runtime (or rather, any USB runtime really, but a DFU runtime would make things "just work") I think we could actually upload firmware to the LVFS.

t-8ch, 2021-03-01:
@Ralim 
I'd like to to take a shot at this (at least for the Pinecil).
The gigadevice firmware library has code to implement the necessary DFU runtime descriptors beginning from version 1.1.0.
Do you think there are any problems updating to the latest version of the firmware?

Ralim, 2021-03-03:
@t-8ch 
You should be totally fine to update to the latest gigadevice library, though we dont _require_ their DFU at runtime as we can use the rom version; so probably just need to setup the barebones USB to enumerate with a descriptor.

Sorta thinking could just implement a CDC or something while at it but unsure what is most useful long term as well.

Totally open to ideas here :)

hughsie, 2021-03-03:
> setup the barebones USB to enumerate with a descriptor

Certainly if you implement a DFU runtime interface then it's easy for fwupd to put the device into bootloader mode without asking the user to reinsert without pressing a button -- i.e. it goes from "urgh" to "just works".

Would there be a different USB PID for each language? Or perhaps a different USB REV? Either works for fwupd, we just need to know the language from the descriptor so we can download (from the LVFS) and then write the correct .bin file to the device.

alvinhochun, 2021-03-03:
I don't like the idea of using different PIDs for different languages. That means assigning a new PID every time a new language is added and having to keep a mapping between PID and language. Wouldn't it be more flexible to provide some mechanism (e.g. HID reports but I don't know what's the best way) for the host software to query the current version and language?

I think once such feature is added, it may also be utilized by the [official Pinecil firmware updater](https://github.com/pine64/pinecil-firmware-updater) tool to make the manual update process more user-friendly.

CC @gamelaster

gamelaster, 2021-03-03:
@alvinhochun as I said, if there will be higher demand of Linux version, it's not problem to create it. Although, I'm bit worried about packaging, since it's best to ship the app with custom dfu-util. I will think about this and try to port it over Linux.

alvinhochun, 2021-03-03:
> 
> 
> @alvinhochun as I said, if there will be higher demand of Linux version, it's not problem to create it. Although, I'm bit worried about packaging, since it's best to ship the app with custom dfu-util. I will think about this and try to port it over Linux.

No, I was mainly referring to the use on Windows and macOS. While this issue is mostly focused on LVFS, the addition of a runtime interface may also allow the updater to automatically detect firmware version and flash the updated firmware without requiring the user to manually enter DFU mode by holding the "-" button.

hughsie, 2021-03-03:
> e.g. HID reports

If different PIDs are not going to fly (to me a product in a different language is a different thing, but that's juts me...) then you could use a different REV for the languages. e.g. REV=0 for en, 1=fr, 2=de etc. If you did either REV or PID then there are no code changes required. If you do want a custom HID interface with some sort of command set to:

* get the language
* switch the device to bootloader DFU mode

then it would also be quite simple to support, something like https://github.com/fwupd/fwupd/blob/master/plugins/system76-launch/fu-system76-launch-device.c would be fine.

The other thing to consider is just adding the language in a string descriptor. e.g. in the serial, product, or a custom index. Then fwupd (or the windows updater) can just read that without claiming any interface.

Ralim, 2021-03-03:
My vote for language would to put its name in a string in the descriptor as that also removes the requirement to have a number <-> name lookup and instead we can just place it into the USB descriptions somewhere.

hughsie, 2021-03-03:
> instead we can just place it into the USB descriptions somewhere

Works for me; it would be a few lines in fwupd to parse that and add an extra GUID.

gamelaster, 2021-03-03:
> No, I was mainly referring to the use on Windows and macOS. While this issue is mostly focused on LVFS, the addition of a runtime interface may also allow the updater to automatically detect [...]

Ah sorry, yeah, this is interesting idea and if this whole feature will be implemented into IronOS and LVFS, I guess it can be implemented into the app.

0Karakurt0, 2025-07-16:
Random internet denizen coming by, is the idea still in place and in progress?
Would love to have such feature!