Ralim/IronOS #820: dfu-util error

HanaO00, 2021-01-21:
Hello,

I followed the guide using the `dfu-util -d 28e9:0189 -a 0 -D Pinecil_FR.bin -s 0x08000000:mass-erase:force` command, but with the `mass-erase:force` option dfu-util doesn't work, it gets there by removing the `mass-erase:force` option, but in the Pine64 wiki it says to use this option, am I the only one having this problem, maybe came across a faulty Pinecil?

Thanks
[image]

Ralim, 2021-01-22:
This is most likely a different dfu-util version compared to what is being used elsewhere.
The mass erase is nice in that it resets settings and is slightly faster but there is no harm in not using it.

HanaO00, 2021-01-22:
Thank you for your answer, I thought it created instabilities not to use mass-erase:force, I will try to find a live debian to see if I have the same error

losinggeneration, 2021-02-01:
> This is most likely a different dfu-util version compared to what is being used elsewhere.

Doesn't *seem* to be version related.

Debian with `dfu-util` installed from apt.
```
sudo dfu-util -l

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [28e9:0189] ver=0100, devnum=10, cfg=1, intf=0, path="1-3", alt=1, name="@Option Bytes  /0x1FFFF800/01*016Be", serial="??"
Found DFU: [28e9:0189] ver=0100, devnum=10, cfg=1, intf=0, path="1-3", alt=0, name="@Internal Flash  /0x08000000/128*001Kg", serial="??"
{15:23:00:harley@dreamcast:~/Source/embedded/pine/pinecil}% sudo dfu-util -d 28e9:0189 -a 0 -D Pinecil_EN.bin -s 0x08000000:mass-erase:force
```
```
sudo dfu-util -d 28e9:0189 -a 0 -D Pinecil_EN.bin -s 0x08000000:mass-erase:force

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 28e9:0189
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuDNLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Performing mass erase, this can take a moment
dfu-util: Error during special command "MASS_ERASE" get_status 
```

dfu-util checked out tag v0.10 
```
sudo dfu-util-0.10 -l

dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [28e9:0189] ver=0100, devnum=10, cfg=1, intf=0, path="1-3", alt=1, name="@Option Bytes  /0x1FFFF800/01*016Be", serial="3TBJ"
Found DFU: [28e9:0189] ver=0100, devnum=10, cfg=1, intf=0, path="1-3", alt=0, name="@Internal Flash  /0x08000000/128*001Kg", serial="3TBJ"
```
```
sudo dfu-util-0.10 -d 28e9:0189 -a 0 -D Pinecil_EN.bin -s 0x08000000:mass-erase:force

dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
[... 40 lines cut ...]
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
    |__ Port 2: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 480M
    |__ Port 2: Dev 3, If 0, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 2: Dev 3, If 1, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 3: Dev 11, If 0, Class=Application Specific Interface, Driver=, 12M   (<====== this one)
    |__ Port 8: Dev 4, If 3, Class=Human Interface Device, Driver=usbhid, 480M
    |__ Port 8: Dev 4, If 1, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 8: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 8: Dev 4, If 0, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 9: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 8, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 3: Dev 9, If 0, Class=Vendor Specific Class, Driver=ti_usb_3410_5052, 12M
```

> The mass erase is nice in that it resets settings and is slightly faster but there is no harm in not using it.

From the [Pinecil wiki](https://wiki.pine64.org/wiki/Pinecil#Manually)
> :mass-erase:force is required to force-erase flash before updating firmware, otherwise you might have unpredictable results.

So if that's not the case, perhaps the Pincecil wiki should be updated to indicate it's optional instead of what's there now  which seems to indicate there's a high chance of  failure without it.

Ralim, 2021-02-02:
So its absolutely not _required_ so long as you are flashing IronOS over IronOS.

I have 3 Pinecil units on hand, from pre-release through to the latest batch and none of them have this issue on my arch systems; which is why i was suspect of it being a dfu-util release related issue.

losinggeneration, 2021-02-02:
> So its absolutely not _required_ so long as you are flashing IronOS over IronOS.

I think that's well worth updating the Pinecil wiki to make clear. I'll try to do that after work today. Hopefully that'll cause less noise here as people potentially run into a similar issue.

> I have 3 Pinecil units on hand, from pre-release through to the latest batch and none of them have this issue on my arch systems; which is why i was suspect of it being a dfu-util release related issue.

I was able to reproduce with both 0.9 (from apt with Debian testing) & 0.10 (git tag) which makes me question if it's dfu-util release related. Looking at the Arch PKGBUILD, the only dependency it has listed is libusb. With that in mind, both of my builds that failed would have been linked to the same version of libusb. That also sort of looks like a dead-end. Both include the exact same descriptor parsing patch.

I may setup either an Arch chroot or container tonight & give it a try to see if it can be further narrowed down to a distro & packaging issue.

DooMMasteR, 2021-02-06:
I have the same issue on Windows as well as on Linux (dfu-util v 0.10 from git).
`
dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 28e9:0189
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Found GD32VF103, which reports a bad page size and count for its internal memory.
Fixed layout based on part number: page size 1024, count 128.
Downloading element to address = 0x00800000, size = 48800
dfu-util: Last page at 0x0080be9f is not writeable
`
is the issue when I try it without the erase command

alvinhochun, 2021-02-10:
>     Found DFU: [28e9:0189] ver=0100, devnum=10, cfg=1, intf=0, path="1-3", alt=0, name="@Internal Flash  /0x08000000/128*001Kg", serial="3TBJ"

I noticed that your output says `ver=0100` and `name="@Internal Flash  /0x08000000/128*001Kg"`, while my devices that works with `mass-erase` produces `ver=1000` and `name="@Internal Flash  /0x08000000/512*002Kg"`. Could this indicate a different DFU bootloader on the chip?

Ralim, 2021-02-10:
This could indicate that Gigadevice has changed the bootloader used in the ROM inside the chip :(

crenn, 2021-02-10:
There has been a few reports of this on the Pine64 forum as well:
https://forum.pine64.org/showthread.php?tid=12803
My own testing and logs are there as well. Eventually did try programming without mass-erase and had no problems with it.

gamelaster, 2021-02-16:
I had same issue in Pinecil Firmware Updater tool.
There are two cases I found so far:
- Pinecils, which needs mass-erase, else the uploaded firmware doesn't work
- Pinecils, which fails on mass-erase, without mass erase, uploading works and the firmware works good too.

So I made in the app this updating optional. Also note, that dfu-util v0.10 have some improvements.

jannic, 2021-03-16:
I did a pcap dump of the usb communication of such a failed mass erase request:
[mass_erase_dump.zip](https://github.com/Ralim/IronOS/files/6150555/mass_erase_dump.zip)

It seems like the pinecil answers with an EPIPE error to the mass erase command. However, I have no idea how the correct answer should look like, and if the error is caused by the pinecil, of if something in the request was wrong.

Strayer, 2021-03-24:
I recently purchased a Pinecil and faced the same issue on my MacBook with macOS Big Sur 11.2.3 (20D91). I'm using dfu-util 0.10 from Homebrew. The error message was: `dfu-util: Error during special command "MASS_ERASE" get_status`

In the end I dared removing `:mass-erase:force` from the flashing command and it seemed to work fine. Debug menu states I'm on v2.14.245902 17-01-21, which seems correct.

HanaO00, 2021-04-30:
I have the impression that dfu-util does it even without setting the erase option
[image]

skorokithakis, 2021-05-11:
I got the same error, it works without `:mass-erace:force` for me as well.

closed 2021-08-06 by Ralim

alnI3S, 2024-09-27:
Hi all,
I have the same issue `mass-erase` with `dfu-util` command on a STM32H7 board.

I've found a solution with the `STM32CubeProgrammer` app under Ubuntu. Here below are the steps to do:
1)  plug the board (boot mode is not nesessary) by USB to the computer. If you don't have a USB plug you can buy a ST-LinkV2 adaptator and connecting pins GND, SWDIO, SWCLK (don't use the 3.3 V if you can power your board by other manner).
2)  in the app, select USB: will dislay Port:USB1
3) press Connect
4) in the menu icon (top-left) click `Erasing & programming`
5) You can use section ` Download > Start Programm...´ or `Full chip erase`
6) when done, disconnect and quit

PR #1025 (2021-08-06): Alternate flash command for Pinecil w/o mass-erase
It seems the common wisdom at this point is to flash without the explicit erase if `dfu-util` is unable to complete the erase.

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes - ie it worked on my Pinecil

* **What kind of change does this PR introduce?**
Documentation-only update.

* **Other information**:
Resolves #820.