Ralim/IronOS #821: Pinecil keeps resetting with USB-PD after connecting breakout board

Depau, 2021-01-21:
_This is a Bug//Question/IDK TBH_

* **I have**
  - [x] Searched previous issues 
  - [x] ~~This is in _this_ firmware, not vendor "offical" firmware~~ this is the vendor firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU) - **I don't think so, but see below**
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

I received today my new Pinecil (second batch), I also got their "PinePower" USB-PD/QC power supply (the bigger one with the screen) and the Pinecil breakout board.

Everything was working when I first tested, then I tried plugging in the breakout board to the iron and to the PD port of the power supply - nothing attached to the board, I just wanted to see if everything was working.

The LED on the breakout started flashing (~0.5s off 0.5s on) and the iron display didn't turn on. The PSU reports sometimes 0V, sometimes 5.1V - the reading seems to refresh roughly every 1s so it's not instantaneous.

I can tell that the board is resetting because after plugging it in, if I press the "-" button at a random time it always goes into DFU mode. I can tell it goes into DFU mode because it stays at 5V and the light stops flashing - just like what happens if I plug it in to a computer.

The breakout board seems to be working fine since if I plug it to a computer it acts as if it weren't there, the LED stays on and the board shows up as DFU in dmesg if I press the "-" button on connection.

I tried updating to the latest **pre-built** release, the issue still occurs. However, I had strange behavior with that too: I was not able to download the firmware with `mass-erase:force`:

<details>
<summary>Command output (click to expand)</summary>

Note: the output is from the latest dfu-util from git, there's no difference with 0.10 stable. I haven't tested 0.9.

```
# dfu-util -d 28e9:0189 -a 0 -D /tmp/Pinecil_EN.bin -s 0x08000000:mass-erase:force

dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 28e9:0189
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Found GD32VF103, which reports a bad page size and count for its internal memory.
Fixed layout based on part number: page size 1024, count 128.
Performing mass erase, this can take a moment
dfu-util: Error during special command "MASS_ERASE" get_status
```

```
# dfu-util -l
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found Runtime: [0a5c:21e8] ver=0112, devnum=3, cfg=1, intf=3, path="1-5", alt=0, name="UNKNOWN", serial="5CF3709B8CFF"
Found DFU: [28e9:0189] ver=0100, devnum=94, cfg=1, intf=0, path="3-1", alt=1, name="@Option Bytes  /0x1FFFF800/01*016Be", serial="3TBJ"
Found DFU: [28e9:0189] ver=0100, devnum=94, cfg=1, intf=0, path="3-1", alt=0, name="@Internal Flash  /0x08000000/128*001Kg", serial="3TBJ"
```

</details>

It worked fine without (`-s 0x08000000`)

<details>
<summary>Command output (click to expand)</summary>

```
# dfu-util -d 28e9:0189 -a 0 -D /tmp/Pinecil_EN.bin -s 0x08000000

dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 28e9:0189
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuDNLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Found GD32VF103, which reports a bad page size and count for its internal memory.
Fixed layout based on part number: page size 1024, count 128.
Downloading element to address = 0x08000000, size = 48800
Erase   	[=========================] 100%        48800 bytes
Erase    done.
Download	[=========================] 100%        48800 bytes
Download done.
File downloaded successfully
```
</details>

I am not aware whether this issue is related to the main issue: I only tried to reflash it *after* I started having the issue.

Also, quick charge works fine and it's able to negotiate up to 12V.

I'm going to solder the pin headers to the breakout and see if I get anything from UART.

* **What is the expected behavior?**

PD works :man_shrugging: ;)

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: v2.14.2425902
  - PCB Version: 17-01-21 sounds more like a build date but here you go
  - Power Supply (Voltage and Current Rating): PinePower 120W, issue only with PD port.

By the way, power delivery is still working and negotiates fine with my laptop and phone, so the PSU is working. I don't have another PD PSU to test.

Depau, 2021-01-21:
I didn't mention it but after the breakout board incident the issue occurs even if connected directly to the PSU.

I also tried to factory reset it from advanced settings, PD is still broken.

Ralim, 2021-01-22:
Alrighty, so this is interesting on a few accounts.

(1) The pd stack in this firmware _never_ sends a hard reset command to a power supply. So it does not request a reset of the power.
(2) The breakout board is a 1:1 straight through on the CC pins.
(3) The firmware cant be changed by the breakout board.

To rule a few things out, have you tried more than one usb-c cable?
My suspicion is with the beginning of the code where the FUSB changes the pullup/down to figure out which CC pin is connected.
If there is a bad contact here that could be enough to push over the edge and make the psu think you unplugged the device.

I have one of those power supplies on my desk, its one of the ones i test on before a release so its definitely working for me here on all of my units.

alvinhochun, 2021-01-22:
Very high chance that it is a bad CC pin connection. If you flip the Type-C connector around it might go away, which will further prove this.

The question is, whether the CC pin connection is broken on the Pinecil or the breakout board. Since you mentioned that it happens even with a direct connection, it looks like to be a problem with the Pinecil board and not the breakout board.

I also have a breakout board which has a bad CC pin connection at the male Type-C connector end. The PD supply resets when plugged in one orientation and the board flexes. I fixed it by reworking the solder joints a bit, which was quite tricky to do.

Depau, 2021-01-22:
Hello,

Thank you all for your answers. This is actually the first device I've ever used with USB-PD - my other devices support it but they have other means of charging. Therefore I don't have any other PD PSUs or C to C cables, but I can definitely buy a new cable, just to be sure.

Though again, the same cable works fine with my laptop and phone.

Flipping the type-c connector does not help, neither does help changing which side of the cable goes to the PSU.

---

> My suspicion is with the beginning of the code where the FUSB changes the pullup/down to figure out which CC pin is connected.
> If there is a bad contact here that could be enough to push over the edge and make the psu think you unplugged the device

and

> I also have a breakout board which has a bad CC pin connection at the male Type-C connector end. The PD supply resets when plugged in one orientation and the board flexes. I fixed it by reworking the solder joints a bit, which was quite tricky to do.

Considering that the type-c ports on both the iron and the breakout are quite stiff, I think these the two snips I quoted might be the most appropriate. It might have busted some of the joints due to the strength it required. I'll definitely try that, I'll let you know how it goes.

Ralim, 2021-01-22:
If your not using of devices then the cc pin may not matter. Though if you are charging a laptop via USB C it should be using pd.

If you had blown the PD chip the firmware would warn you; so we can be reasonably sure that the if itself is "ok".

Most likely is connector damage unfortunately :'( 

May be worth trying wiggling cable around or looking closely at the port.

Cable should have a tad of movement but shouldn't feel "loose"

Depau, 2021-01-22:
Alright, I did some troubleshooting with a cheap aliexpress scope. I taught myself the basics of PD negotiation,  I did some testing on the breakout connected to different devices and I compared the results

This is what I see on the between Vbus and ground:
[image]

This is what I see between CC and ground (Vmax is 1.7V but keep in mind the scope is garbage):

[image]

Then I noticed that indeed orientation does matter: if I flip the connector, I get 0V. So one of the CC pins is definitely shorted to ground on the iron side. However, I don't get why it keeps resetting when I plug it the other way. Shouldn't it work if it's only the other CC pin that is shorted to ground?

alvinhochun, 2021-01-22:
> I then noticed that indeed orientation does matter: if I flip the connector, I get 0V. So one of the CC pins is definitely shorted to ground on the iron side. However, I don't get why it keeps resetting when I plug it the other way.
> 
> Shouldn't it still work if it's only the other CC pin that is shorted to ground - i.e. the cable is flipped the other way -, as alvinhochun suggested earlier? I think there is not just one short.

Hold on, the CC pins are pulled down to GND via a 5.1k ohm resistor (IIRC) on the UFP (Pinecil end), not shorted. Typical PD supply side only has one of the CC pins connected (again, IIRC) so it is completely normal for one of the CC pins to be at 0V. The PD negotiation happens on the other connected CC line. Which CC pin is connected depends on the cable orientation.

So you need to probe both CC pins and also rotate the orientation of one end of the cable to get the full picture. (Rotate only one end -- if you rotate both ends the same CC pin ends up being connected.)

Depau, 2021-01-22:
> > I then noticed that indeed orientation does matter: if I flip the connector, I get 0V. So one of the CC pins is definitely shorted to ground on the iron side. However, I don't get why it keeps resetting when I plug it the other way.
> > Shouldn't it still work if it's only the other CC pin that is shorted to ground - i.e. the cable is flipped the other way -, as alvinhochun suggested earlier? I think there is not just one short.
> 
> Hold on, the CC pins are pulled down to GND via a 5.1k ohm resistor (IIRC) on the UFP (Pinecil end), not shorted. Typical PD supplies with an integrated cable only has one of the CC pins connected (again, IIRC, ~but I'm not sure how this applies to a supply without an integrated cable; I got confused when typing this reply~ I think normal passive cables also only has one CC wire?) so it is completely normal for one of the CC pins to be at 0V. The PD negotiation happens on the other connected CC line. Which CC pin is connected depends on the cable orientation.
> 
> So you need to probe both CC pins and also rotate the orientation of the two ends of the cable in different combinations to get the full picture.

That's what I did, I wasn't clear enough.

R10/R11 below refer to the 0 ohm resistors placed in the CC lines in the breakout board, I'll refer to them instead of the lines they're connected to to avoid further confusion. R10 is connected to CC1, R11 to CC2, in any case, at least according to the schematic.

The cable orientation is kept consistent relative to the breakout board in the measurements below, I instead rotate the iron.

Measurements with breakout board with the LED facing **UP**, iron with the display facing **UP**:
- Connected to the PSU - LED flashes:
  - R10-GND: 0V
  - R11-GND: 1.7V but with the waveform I posted earlier

- Not connected to PSU
  - R10-GND: 0 ohm
  - R11-GND: ~900 ohm

- Connected only to PSU:
  - R10-GND: 0V
  - R11-GND: ~4V, stable

Breakout board with LED facing **UP**, iron display facing **DOWN**
- Connected to the PSU - LED does NOT flash:
  - R10-GND: 0V
  - R11-GND: 0V

- Breakout connected only to the pinecil:
  - R10-GND: n/c (wtf :thinking:)
  - R11-GND: 0 ohm

**Now the cable is flipped relative to the breakout, and the cable is kept with the same orientation**

Breakout board with the LED facing **UP**, iron with the display facing **UP**:
- Connected only to PSU:
  - R10-GND: ~4V, stable
  - R11-GND: 0V

- Connected to the PSU - LED does NOT flash:
  - R10-GND: 0V
  - R11-GND: 0V

Breakout board with LED facing **UP**, iron display facing **DOWN**

- Connected to the PSU - LED flashes:
  - R10-GND: 1.7V but with the waveform I posted earlier
  - R11-GND: 0V

Depau, 2021-01-22:
Alright, things got a lot worse. I *thought* quick charge was still working. I never tried to heat it up after the initial testing. The moment I pressed the button I got some very nice magic smoke and the iron is now fried.

Thank you guys, you were very helpful :) Also thumbs up for the firmware ;)

closed 2021-01-22 by Depau

Ralim, 2021-01-23:
Well thats honestly terrible.
Make a ticket with Pine; they will usually do what they can to help.
Feels like a manufacturing defect to me.

alvinhochun, 2021-01-23:
That sucks. Hopefully PINE64 should be able to send you a replacement.

I'd be interested to see what's left on the board. Would you mind opening it up and take some photos for our amusement?

A side note: I thought you can measure the 5.1k ohm on the CC lines when not connected to a power source, but no, the FUSB302 states that it needs to be powered to provide that resistance, and just probing when unpowered gives meaningless measurements. Now I'm really confused as to how it triggers the power source to start supplying VBUS.

Ralim, 2021-01-23:
I would also love any photos of the inside if you can, but not required.

I believe the FUSB has a weak internal pulldown on the pins until it is powered and then it turns them off once powered.

Depau, 2021-01-23:
Pine64 already confirmed they'll send me a new Pinecil body.
I'll send some pics of the body later. I haven't opened it up yet but I'm
hoping i can at least remove the USB port completely and use it with the
barrel jack, as a backup iron.
Hopefully the PCB is still intact.

Depau, 2021-01-23:
*of the board

Depau, 2021-01-23:
Here are some pics of the PCB, sorry for the not-so-high quality, this is the best my phone could do.

[image] [image] [image]<br/><br/>

The smoke appears to shoot out of the 5th-6th pin from the bottom, which indeed appears to be CC2. And in fact they don't look right.

When plugging in the PD cable in the orientation that earlier caused the LED to blink, now that pin gets red hot and burns the plastic above it.

If I flip the connector it doesn't smoke but it doesn't turn on either (scenario "display DOWN breakout UP"). With QC3 it now smokes in both orientations.

Anyway, Pine64 wants me to send it back, I don't think that's gonna happen since the cheapest shipping option I could find costs more than a new iron incl. shipping :/