Ralim/IronOS #824: Releases should include copyright and licence notices

alvinhochun, 2021-01-23:
_This is a [~~Bug~~/~~Feature Request~~/~~Question~~/Complaint]_

* **I have**
  - Searched previous issues 
  - This is in _this_ firmware, not vendor "offical" firmware
  - ~~This is not a bug in the vendor bootloader (aka DFU)~~ (not applicable)
  - ~~I have checked this is not already covered in the docs in `/Documentation`~~ (not applicable)

* **What is the current behavior?**

(not applicable)

* **What is the expected behavior?**

(not applicable)

***Steps to reproduce the bug:***

(not applicable)

* **What is the motivation / use case for changing the behavior?**

The copyright notices are needed to comply with the software licences.

- The firmware release itself is GPLv3, therefore the GPLv3 copyright notice should be included.
- FreeRTOS used in the firmware is MIT license, therefore the MIT license copyright notice should be included.

I think putting them in a text file inside the release ZIP archives should be sufficient. This should also apply to the GitHub Action artifacts.

* **What are you running:**

(not applicable)

Vinigas, 2021-01-23:
Isn't it applicable to the source code, not the compiled artifact itself ?

alvinhochun, 2021-01-23:
> Isn't it applicable to the source code only, not to the compilation artifacts ?

I am not a lawyer, but general consensus is that the binaries compiled from source are considered "copies of the software".

And when it comes to GPL, there is no question that the compiled binaries are derivative work of the source code.

Ralim, 2021-01-24:
I'm happy for this to be done; at the moment github is not charging for the size / storage of the artefacts so im not worried by this.
If anyway wants to make a PR to automatically put these in the exports I would love it; if not I can "soon".

alvinhochun, 2021-02-03:
It looks like there are some other dependencies that uses the Apache 2.0 license, like the NMSIS code and the FUSB302 driver, so need to keep note of them too.

And then there are odd ones that I don't know how to deal with, like:

- https://github.com/Ralim/IronOS/blob/31cbbc4489b239dc744fd8cb24a4d257cf5dad53/source/Core/BSP/Miniware/FreeRTOSConfig.h - GPLv2 with FreeRTOS exception, is this from before FreeRTOS changed to MIT license?
- https://github.com/Ralim/IronOS/blob/31cbbc4489b239dc744fd8cb24a4d257cf5dad53/source/Core/BSP/Pine64/Vendor/SoC/gd32vf103/Common/Include/nuclei_sdk_soc.h and some other files - only states "see LICENSE for license details"

PR #844 (2021-02-06): Add copyright and license notices to release
This adds `LICENSE` and a new `LICENSE_RELEASE.md` to the GitHub action workflow artifact outputs.

@Ralim  I think this should be appropriate, but if you find any issues, feel free to edit the files before merging

Closes #824
closed 2021-02-06 by Ralim