Ralim/IronOS #827: Temperature in idle is 220 + degrees

Gwebber-Net, 2021-01-24:
This is a bug.

* **I have**
  - [x] Searched previous issues 
  - [x] This is in latest release as of now.  v2.14.1  2425902 
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
      Temperate in idle is around 220 degrees celsius.
* **What is the expected behavior?**
      Temperature should be around 22 degrees celsius
***Steps to reproduce the bug:***
 1 of my pinecils has this bug, the other one does not.

* **What is the motivation / use case for changing the behavior?**
 
 ?

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.14.2425902
  - PCB Version: 3
  - Power Supply   4s Lithium pack or 12V usb C (both power supplies generate the issue)

Gwebber-Net, 2021-01-24:
Debug info from my properly working Pinecil
hw g 150
hw m 134
hw p 50
rtip 690
ctip 23
chan 235
pcb 3
pwr 1
max 405

Debug info about the not properly functioning pinecil
HW g 90
HW m 134
HW p 46
Rtip 720-730
ctip 226
chan 250
vin 12.1 (wich is correct)
pcb 3
pwr 1
max 606

Ralim, 2021-01-25:
* Have you done a configuration reset on the unit?
* Can you double check `chan` as for both units you should be ctip == chan/10  as your less than the default cal value.

Gwebber-Net, 2021-01-27:
Yes i have done a factory reset inside the pinecil itself, ive reprogrammed it a couple of times aswell.

Properly working pinecil 
Chan : 208

Badly working pinecil
Chan 220

Ralim, 2021-01-28:
Can you confirm that you have **not** done the calibration step?

The maths for converting a tip temperature to a handle temperature is fairly dumb and should not spit out a temp like this.

If you heat the iron to say 300C what does it do with the temperature readings?

discip, 2022-04-21:
@Gwebber-Net
Is this issue still present?
If not, please don't forget to close it. 😊

closed 2022-05-08 by Ralim