Ralim/IronOS #837: [Feature Request] USB configuration protocol

fonix232, 2021-01-30:
_This is a [Feature Request/Question]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

IronOS provides no way to fine-tune the settings from a connected PC.

* **What is the expected behavior?**

A custom USB protocol being present for configuration and monitoring.

* **What is the motivation / use case for changing the behavior?**

Simplifying configuration of the soldering iron with ease, possibly monitoring

---

First of all, a gigantic thank you for this project. I've only had my TS80P for a few days, and used it both on the official firmware and the latest IronOS release. Latter is visibly more reliable and stable, not to mention the relief caused by not depending on (currently) closed source solutions and the manufacturer's whims for future support. It's great to have open source software for practically everything in life, let it be a light switch, a soldering iron, or an e-cigarette (this will be relevant in a moment).

One thing I found a bit lacking is the configuration. Don't get me wrong, the menu system is great and straightforward, but cumbersome to set up if you have to change a lot of things, and due to the screen size limit, it's hard to show every relevant bit of information.

The readme does specify that IronOS does not support the official firmware's approach of using UMS for configuration. I completely agree with this - that solution (including the DFU mode) is simplistic, backwards, and has a lot of potential issues. UMS was never meant to be used like this. 

However, I think that configuration via a connected computer/phone/smart fridge should still be possible. There is in fact a very similar, albeit non-OSS project, a custom firmware for electronic cigarettes/vapes. And if you think about it... Vapes and soldering irons have a lot in common on the low level. Both of them are precision heating systems where you take the input power, and push it to a heating element when needed. Both of them are about the precise timing of the supplied power to maintain the temperature, and they even work on similar power levels (most vapes supported by the project are high power ones, 120-200W models). The main difference in hardware is probably the battery circuit in vapes that takes care of charging and protecting the cells. Obviously since these soldering irons work from mains (or high power powerbanks), this is not needed here.

The project in question is called [NFE](https://nfeteam.org/forum/) with two distinct products: ArcticFox (for monochrome display devices) and RedPanda (for colour LCD units). But what both of these have in common is a standardized (but closed) protocol over USB that allows the supplied configuration tool to change settings on the vape, create profiles, and even themes/custom UI/firmware update. It also allows for carrying over settings to a new vape with ease, and they even built a monitor into it.

ArcticFox in fact runs on the very same platforms these soldering irons do - STM32 and Nuvoton MCUs. So in theory it should be possible to create this custom protocol for IronOS too. Unfortunately I can't offer much on the IronOS side of things - I'm no microcontroller engineer, and I can't C much. However I do have some experience with both .Net and Kotlin/Native, both of which could be used to write a cross-platform app for the host machine for the configuration purposes. 

I'm not sure if the monitor functionality is viable in any way - I don't know any laptops or PCs that can provide the necessary 12V or 20V USB-PD or QC2.0/3.0 power output while the data lines are connected, and unlike previous USB protocols, cable splitting is not really an option, since the soldering iron needs to negotiate with the power supply. The configuration (and maybe even UI editing?) functionality should be achievable though - unless I'm missing a detail, in which case please do correct me.

All in all I think it would be a neat idea if IronOS allowed configuration through an external connected device. It would make setup easier, and users are usually more comfortable with a nice big UI than fiddling through a menu on a small screen with only two buttons.

Ralim, 2021-01-30:
I'm with you on this being a good feature request.

Main issues here are:
* Code size (TS80 has very tiny amount of flash left)
* Closed source protocol is no-go, but implementing an open one I'm open to.

This would be entirely a doable thing on the Pinecil as we still have around half the flash spare on it. On the Miniware irons however, we would need some serious cleanup and optimisation of the code to shrink it to fit a "full" usb stack.

Monitoring would be doable with PD as you can split the CC lines away from the rest to split that up. But QC would be a no-go or naturally with DC input (TS100/Pinecil).

This said, I've just done a basic pass of a uart for the Pinecil in #830  and will most likely build up some form of USB device for it that I'm sure ultimately will allow configuration _somehow_. Exactly how that is done is a good question.

Adding support to this for miniware irons would either involve:
* Getting all of the versions of their bootloaders to reliably program the out-of-spec extra flash that tends to be there
* Shrinking existing code size by 4-10k

fonix232, 2021-02-02:
> Closed source protocol is no-go, but implementing an open one I'm open to.

I completely agree - the reason I linked the NFE site is purely for reference, i.e. proof that it can be done.

> This would be entirely a doable thing on the Pinecil as we still have around half the flash spare on it. On the Miniware irons however, we would need some serious cleanup and optimisation of the code to shrink it to fit a "full" usb stack.

Do we need a full USB stack here? I'm not attached to having a full-blown USB protocol - as long as a connected device (let it be a laptop, desktop PC, smartphone, you name it, anything that can handle said communication) can open the channel and communicate with the iron would work. 

> Monitoring would be doable with PD as you can split the CC lines away from the rest to split that up. But QC would be a no-go or naturally with DC input (TS100/Pinecil).

Hmm, I'm yet to see a USB-C PD capable cable that is split between power/data. I remember those old clunky external HDDs that needed two USB ports for power, but I haven't seen anything remotely similar for USB-C - and in my opinion, DIY'ing such a large power throughput thing is not wise for most people. I understand that people using a programmable soldering iron would most likely have the skill, but it's a dangerous thing nonetheless.

> This said, I've just done a basic pass of a uart for the Pinecil in #830 and will most likely build up some form of USB device for it that I'm sure ultimately will allow configuration _somehow_. Exactly how that is done is a good question.

I believe the NFE team also used UART over USB for the configuration exchange. As for the how... I'm fairly certain I have even less ideas than you.

> * Getting all of the versions of their bootloaders to reliably program the out-of-spec extra flash that tends to be there

If I understand you correctly, this would involve using a custom bootloader/DFU on these irons, right? The flashing guide links to dapboot, however as far as I can see it's not as straightforward to flash - plus the TS80/TS80P are not supported, and dapboot hasn't been updated for these devices for 3 years now (even though the mainline dapboot project has seen some recent updates). On the other hand, using dapboot would be beneficial (easier updates due to the proper DFU implementation, and even WebUSB - making it a simple "click here to flash" website for updates, not unlike the flasher website Google has made for their Pixel devices).

> * Shrinking existing code size by 4-10k

Knowing almost nothing about C++ and the codebase, this sounds like the "easier for the user, much harder for the developer" approach.

zeroping, 2022-06-10:
I'll leave my $0.02 here.

I'm using a Pinecil, and I am finding is good enough to replace normal base-station soldering irons. The thought had occurred to me: would it be possible to build an iron holder / base station with a big display and tactile knob that could control the temperature on a connected Pinecil? I'm envisioning something to compete with the fanciest bench-top soldering irons. Something that would pass through USB PD from a PD power supply, but would connect to the Pinecil over USB for control. That led me here.

It sounds like a hacky version of this would already be possible using UART, but it would be awesome to do this with the Pinecil as a USB device. There's a few microcontroller-sized things that could negotiate as USB host to be the heart of a 'base station'.  An initial starting point could be implementing USB logging and control of the iron from a simple command-line tool.

Anyways, lots of overlap with the efforts to do external telemetry and logging. Just to say it, it _does_ look like there are "USB Type C Splitter" devices on Amazon that split the USB PD lines and the USB (2.0) data lines, but a test is needed to really know.

Just to say it, it looks like acting as a [CDC serial USB device](https://github.com/linusreM/GD32V-Virtual-COM-Port) shouldn't be _too_ hard for the Pinecil. USB HID probably makes more sense, but I don't see a nice demo for that. I know the STM32-based hardware can do this too, but it sounds like the situation on the flash-constrained devices might be trickier. 

tl;dr: A USB-connected 'smart' soldering iron base station is one use case for this feature.

Ralim, 2022-06-27:
Hia,

Yes I agree that acting as a CDC serial device would not be too hard; just _someone_ needs to write the code for that, and figure out the best way to handle muxing that with QC protocol. (I just don't really have time at this point in time to implement USB for the Pinecil)

My preference is to lean towards uart or simpler communication protocols as its most likely the device that would want to talk to the Pinecil could be made far easier if usb was not involved. And we have a UART already broken out.

If someone contributes a USB stack I would happily merge it and look after it, but I just dont have time to get one going in the foreseeable future.

zeroping, 2022-06-27:
Ralim: Not saying you personally should prioritize, I'm just glad to see the idea on the ideas list for the project as a whole. You certainly do enough for the project as it is!

I like the idea of USB on the Pinecil, since it is more modular/standardized than using a UART on some USB-C pins in a non-standard way. I'd love to see the Pinecil speak USB CDC serial with the same protocol as all the irons (currently?) speak via UART. It gives multiple routes to get the same functionality.

(I'd wondered if it would be better to use USB HID, but compatibility with existing UART control seems preferable.)

I took a first run at getting some of the USB CDC demos running on my Pinecil, and haven't gotten very far yet. It's hard to tell which demo code is most likely to play nicely with the IronOS code. But I haven't given up yet. 
(if there's a good community in which to chat and ask for help, let me know)

I also tried to obtain an appropriate splitter cable from Amazon, and haven't gotten a working one yet. I think it needs to pass GND, VCC, CC1, and CC2 to one connector for a USB-PD supply to negotiate and provide power, and then needs to pass GND, USB_D+ and USB_D- to another connector for a host to be able to talk USB to the Pinecil. I assume we'll see USB-C hubs that can handle this too at some point.

I need to just get some breakout boards and solder this together to make sure it actually works the way I think it does. 

The good news is that a Pinecil runs fine plugged directly into a laptop, just don't ask it to actually heat up. Might be a nice way to plug in for adjusting config values, then swap to a USB PD supply for soldering.

River-b, 2023-02-23:
@zeroping  you can control your pinecil now from a phone or a PC browser  if you have Pinecil V2 which has BLE in it. 
I'm testing both BLE apps now and I can control my temperature on the phone app or PC screen.
Check out   https://github.com/builder555/PineSAM  (which is getting a lot more features in the next week like live view of temperature)

and check out  [PineSAM  ](https://github.com/builder555/PineSAM )BLE app for Pinecil V2 (PineSAM = Settings and menus, and will have live Temperature view and a Work View screen possibly by the end of the weekend). 
 and also[ Joric's BLE](https://github.com/joric/pinecil) webpage for V2. 
 
1. PineSAM works on windows/linux/ mac any browser (requires python to be installed locally).  will also work on Android or Apple phones or ipads.
2.  Joric's BLE webpage works only in Windows/Linux - no apple devices since they don't support BLE GATT.  and can work in Adroid phones. nothing locally needs to be installed. 
I have personally tested both apps and in phones too. it's nice to have phone horizontal display while soldering and just control it from there. 

note you will need the BLE beta version of V2,  which  there is one in actions right now, just get newest action should work or wait until 2.21 stable is released.