Ralim/IronOS #838: Pinecil repeatedly reseting when using Apple 30W charger

rroohhh, 2021-01-30:
When trying to use the Pinecil with a Apple 30W usb-c charger, the charger seems to constantly cycle the power (the power led on the breakout board is blinking about two times per second). When pressing the button to get into the dfu bootloader, the power seems to stay for a bit longer, but also get turned of again after  about two to three seconds.

Very rarely (after leaving it on the charger for several minutes) the screen on the iron actually comes up and shows it negotiated 15 volts (that seems to reasonable, as the 30W charger only does 20V at 1.5A, which is below the 2.666A wanted by the firmware), pressing the `+` button even starts to heat up the tip, but it resets after two to three seconds aswell (regardless of pressing the `+` button or not).
Getting the screen to come up and getting power for a couple of seconds happens a lot more often with this change:

```patch
diff --git a/source/Core/Threads/POWThread.cpp b/source/Core/Threads/POWThread.cpp
index 2f5661e..78d7df6 100644
--- a/source/Core/Threads/POWThread.cpp
+++ b/source/Core/Threads/POWThread.cpp
@@ -18,6 +18,7 @@

 void startPOWTask(void const *argument __unused) {
   postRToSInit();
+  power_check();^M
   for (;;) {
     osDelay(TICKS_100MS); // Slow down update rate
     power_check();
```

but it still resets after a couple of seconds.

The pinecil is working without problems with other chargers.

The charger is rated for:
- 20V at 1.5A
- 15V at 2A
- 9V at 3A
and I am running commit 5e372310cdb3a5c5b1ccc425831a44ae3b243d71.
This behaviour is occuring regardless of the negotiated voltage (I changed the PD table to make it select 20V or 9V instead of the 15V and nothing changed).

Any ideas what the problem could be, or anything else I can try to help debug / fix this problem?

feiltom, 2021-02-01:
Same Issue with 96W macbook pro Charger

rate for

- 20.5v at 4.7A
- 15v at 3A
- 9v at 3A
- 5.2v at 3A

Ralim, 2021-02-07:
If you are able to do some testing, It would be interesting to test what happens if you _disable_ the QC3.0 code.

I'm wondering if these chargers check the USB2.0 lines and a freaking out when they start to change.

The firmware does _not_ support sending a USB PD reset so the iron itself will not request a power cycle. So I'm slightly suspect of if the QC is causing an issue.

I only have acess to one older 60W (or is it 65W) charger at my workplace and on that charger the unit works completely fine.

rroohhh, 2021-02-07:
@Ralim 
I tried the following:
```patch
diff --git a/source/Core/BSP/Pine64/Model_Config.h b/source/Core/BSP/Pine64/Model_Config.h
index d0ac1e3..8a2c137 100644
--- a/source/Core/BSP/Pine64/Model_Config.h
+++ b/source/Core/BSP/Pine64/Model_Config.h
@@ -17,10 +17,10 @@

 #ifdef MODEL_Pinecil
 #define POW_PD
-#define POW_QC
+// #define POW_QC
 #define POW_DC
-#define POW_QC_20V
-#define ENABLE_QC2
+// #define POW_QC_20V
+// #define ENABLE_QC2
 #define TEMP_TMP36
 #define ACCEL_BMA
 #define ACCEL_SC7
```

And that did not change anything. It sometimes boots up, and shows it negotiated 15 volts, only to reset after a couple of seconds.

Is anything else needed to get the QC3.0 code completely disabled?

DooMMasteR, 2021-02-08:
I have the same issue with an Anker PowerPort+ 5.
USB PD Output: 5V=3A/9V=3A, 15V=2A/20V=1.5A

Ralim, 2021-02-11:
@rroohhh 
That is all that should be required to disable QC so that rules that out.
You could try putting a delay in before PD initalises (are we timing out too fast) ?

@DooMMasteR 
>Edit: I had a look at the source code and there seem to be some hard coded voltage/current stages that need to be met in order for a profile to be selected.
This does not consider the actual possible power over the resistance of the tip.
Since 2666 mA are needed at 20V for the profile to be selected the firmware instead selects 15V which meet the required 2000mA but then, due to the tip resistance, only pulls ~1600 mA.

Yes these are hard coded based on tip resistance for _peak_ current draw.
The average is lower as the tip is always PWM'ed (as we need an off cycle for tip temperature measurement).
We have to size the power adapter on peak power as some PD supplies will "trip off" on over currents even if they are only ~100mS. They also have to be sized for worst case tip performance.
Therefore; all of these are V/R calculations on tip nominal resistance.
When checking power draw ensure you are using a scope or something that can catch the high peak at the start of each on cycle during heat up.

DooMMasteR, 2021-02-12:
Ok, but that should still allow operation at 15V - 2A on the Anker power supply...

rroohhh, 2021-02-13:
@Ralim I tried adding a additional one second delay before initializing the PB parts:
```
--- a/source/Core/Threads/POWThread.cpp
+++ b/source/Core/Threads/POWThread.cpp
@@ -17,6 +17,7 @@
 // Small worker thread to handle power (mostly QC) related steps

 void startPOWTask(void const *argument __unused) {
+  osDelay(TICKS_100MS * 10);
   postRToSInit();
   for (;;) {
     osDelay(TICKS_100MS); // Slow down update rate
```

And this does not really change the resetting. But now it starts of displaying 5 volts in the menu and after about one second it switches to 15 volt. (Which is expected I guess as this delays the PD negotiation by one second. Shortly after it resets just to boot up and do the same again.

Quickly going into the debug menu also confirms a value of `2` for the `PWR` entry.

Ralim, 2021-02-17:
Well that helps us know that the PD negotiation does _complete_ and then something causes the PSU to trip off.
Could you confirm that this issue still occurs with the power pulses disabled?

rroohhh, 2021-02-17:
Disabeling via the menu doesn't change anything.

Ralim, 2021-04-25:
@rroohhh Have you tried the updated code in #917 ?

rroohhh, 2021-04-26:
@Ralim  Thank you for the heads up, I did not try it yet. Ill try it when I get home today.

Ralim, 2021-05-05:
@rroohhh 
Please see newer updated test ones in #949

mmiller7, 2021-09-24:
Ended up here because my new PINECIL is flickering/rebooting with most of the power supplies I own.  All my gadgets are USB-C PD so I only have one QC-compatible charger (which it seems to work on).
-Anker 45W PowerPort+ 5 USB-C PD (5/9/15/20V) -> infinite on/negotiate 15V/reboot
-RAVPOWER 30W power-bank USB-C PD RP-PB058 (5/9/15/20V) -> flickers erratically then eventually negotiates 15V and stabilizes
-BATTPOWER 60W USB-C PD wall charger (5/9/12/20V) -> just works (negotiates 20V)
-Generic 45W USB-C PD car charger (5/9/12/15/20V) -> just works (negotiates 15V)

Ralim, 2021-09-24:
Hello @mmiller7 

To confirm, have you updated to the latest firmware available?

mmiller7, 2021-09-24:
> Hello @mmiller7
> 
> To confirm, have you updated to the latest firmware available?

I think so?  It looks like what it has is newer than what the Wiki mentions, but I'm having a hard time finding information.
When I get it to power up and hold the "-" it shows 16-01-21 v2.14.A17B3C7

Ralim, 2021-09-24:
Ah that is old indeed,
The latest releases are always at the releases page here on github : https://github.com/Ralim/IronOS/releases
v2.15 has much better PD support than the initial v2.14 release

mmiller7, 2021-09-24:
Ah, as soon as I decipher how to do the update I'll give that a try.

Ralim, 2021-09-24:
(1) Download the zip from the release and extract the files
(2) https://github.com/Ralim/IronOS/blob/master/Documentation/Flashing.md#pinecil-pine64

mmiller7, 2021-09-24:
Yeah, both suggested commands fail.

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: Cannot open DFU device 28e9:0189
dfu-util: No DFU capable USB device available

But I can see it exists in 'lsusb' (based on unplugging and plugging in comparing the new ID).  Running Mint 20 (Ubuntu 20.04) Linux.

Ralim, 2021-09-24:
> dfu-util: Cannot open DFU device 28e9:0189

Most likely you will need to run with sudo, as permissions are failing

mmiller7, 2021-09-24:
Ah, that worked with the second command.  I didn't realize just using a USB device would need sudo.

It now blinks several times on my SlimQ dual QC/PD power supply (which "just worked" previously) but eventually works.  Seems much more stable on everything else now.

I'm glad to see work on a PD compatible iron progressing - that's the main thing that has stopped me from getting some of the others out there (since all my other devices are PD-Only and nearly all my supplies are PD-Only)

Thanks!

Ralim, 2021-09-24:
Linux normally locks down any device unless asked not to 🙃  There are udev rules you can use to work around this, but sudo is easier for once-off's

If all PD devices could all be similar would make life a lot easier.

mmiller7, 2021-09-24:
Yeah, I just expected it to be more like flashing Arduinos (serial) or handheld radios (HID) that "just work" once connected.  The error about "invalid" threw me off since it didn't say permission denied I thought it was somehow not connected.

squee666, 2021-12-02:
> Ah that is old indeed, The latest releases are always at the releases page here on github : https://github.com/Ralim/IronOS/releases v2.15 has much better PD support than the initial v2.14 release

Funny update
i hadn't updated my Pinecil since 2nd ever firmware release

i got a new charger that does 65W peak

Type C PD(PPS) 5v/3a 9v/3a 12v/3a 15v/3a 20v/3.25a 

USB A (QC3) 22.5W max

heres the thing i tried it initially on a ts100 with a slightly newer firmware and it would show power and once it ramped up over 300C it would restart
(usb C to dc converter)

the Pinecil on the older firmware would just ramp up to like 410C fine and sit around that

updated the pinecil today to newest firmware since i was many revisions behind and saw USB PD changes

now using this 65w charger it turns on ramps up really fast and soon at it hits over 300c it restarts

Ralim, 2021-12-02:
The most likely cause for this is the new charger not liking the 100% -> 0% transition that happens when the charger reads the setpoint.
First thing I would try is to set the average power limiter to 50W to see how it performs

squee666, 2021-12-02:
> The most likely cause for this is the new charger not liking the 100% -> 0% transition that happens when the charger reads the setpoint. First thing I would try is to set the average power limiter to 50W to see how it performs.

yeah i just realised after some more testing
it seems to be the aggressive wattage dip that make the charger shutoff

every i tried most of my tips TS-I's TS-K's TS-B's
all did same thing every time it hit the temperature margin it would shutoff instantly

only my TS-C2 i had didnt cutoff unless i lowered the temperature

my charger bottlenecks with 48W due to not being higher voltage just higher amps that the pinecil isnt drawing

i tried limiting to 40W but that was making it cutoff sooner weirdly enough

maybe you can stop it ramping down to 0W instantly or something to improve compatibility with future chargers 
since gallium chargers/ 60w+ usb C charger have become more abundant than ever

Ralim, 2021-12-02:
So limiting the ramp down is super messy since that can cause issues with the temperature control.
Buying chargers with proper regulation is often the easier solution, I have 5 GaN 65W units here and none of them exhibit this behaviour. I'm suspect this may be due to poor filtering on the charger side or perhaps a poorly made USB-C cable letting the EMI from the power affect the CC line.

All of the tips will be fairly similar as this comes down to the pid reacting at around 10Hz or so; so its often a fairly sharp cutoff at the end as we often slightly overshoot.

Pinecil sadly cant draw the higher current as it doesnt have an integrated DC/DC (due to cost).

The limit only limits the average by limiting the max PWM duty cycle, if the charger doesnt like the pulsing power then this _may_ be worse as its more choppy (just with lower average). So its worth a test usually.

Could you by chance share what charger and USB C cable you are using? (Dont think I can magic anything from the info but its suuper useful in spotting trends)

squee666, 2021-12-02:
> So limiting the ramp down is super messy since that can cause issues with the temperature control. Buying chargers with proper regulation is often the easier solution, I have 5 GaN 65W units here [...]

I'm using a Kuula 65W gan charger off AliExpress seems to be based of the baseus version.

Outside of using it for the pinecil it's been perfectly fine for what I need it for (PPS)

I tried using a Ugreen 60w USB c cable but that's 2m so that doesn't help and my power bank cable is probably only rated for 45w.

I do have a 100w cable on way.

Suppose this may be the drawback of a cheap charger (I own the 33w version aswell)

Cost me respectively £4+ and £10+ on 11/11 sale Vs paying near £40 for an anker

closed 2022-05-08 by Ralim