Ralim/IronOS #839: Updated firmware, is not heating the tip (tryed new ones, same issue)

Komayo, 2021-02-01:
_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [*] Searched previous issues 
  - [*] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
* ts100 dont heat up the tip after firmware upgrade, i had version from 2019 installed.

* **What is the expected behavior?**
* heating up, now its very hard to even melt the solder, event at 400ºC

***Steps to reproduce the bug:***
Turn on soldering mode,
Soldering tip should  heat up and melt solder
Not enough heat, even with 400ºC no melting
Seems the iron stopped heating after firmware upgrade, even at flux paste box, very hard to heat up the flux(previous firmware, the flux boiled up due to temperature, now only melts and with dificult).

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**
On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.14.2425902
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) : 24V

Ralim, 2021-02-07:
Please flash the vendor firmware first to confirm your hardware is functioning first and also perform a settings reset :)

Komayo, 2021-02-07:
Done that already, working perfectly fine.
Cant install ironos anymore, ts100 dont heat up with it. All works good,
except heating up to the temp set.

Komayo, 2021-02-10:
Today i tryed again the firmware, and noticed something.
I set the soldering temp. for 360º and the TS100 started heating up. Then i noticed the watts are up and down, for example:

0,25 W to 20.0 W then started going down 20, 18, 15.... 5, 2, 0.50 , etc and looping that way.
Seems that is the issue with the tip not heating up, what can i do to try to fix this? Is is my DC power supply, it is a regulated DC from 1V to 32V, 3A max, and i have it on 24DC for the TS100 power.

Ralim, 2021-04-25:
This is hard to debug as there isnt much that would normally cause this other than either a power supply not working well, bad contacts or misbehaving tip.

Could you list out all the values in the debug screen with a tip connected at room temperature?

Komayo, 2021-04-25:
I am working with the official firmware no issues. Working 100%. I had a previous IronOS version also working good, but i cant remember the version number. With the last versions of IronOS(tryed a few), same issue. Start heating up, but dont reach the temperatures set. Cant figure out the issue, i tryed diferent tips, diferrent PSU, nothing fixes. Had to downgrade to official firmware.

discip, 2022-03-31:
@Komayo
Have you tried the latest commit yet?
Should work much better now, but if your tip is bad this will not help either. Had a bad tip as well, temperature was jumping around similar to what you described.