Ralim/IronOS #840: [TS80P] Difficulties with USB PD in current (v 2.14.1) firmware

elschopi, 2021-02-02:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug?]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
TS80P with latest IronOS (v 2.14.1) - when using a Anker PowerPort Speed PD 30 (30W USB PD wall plug), the unit does not negotiate PD power, voltage as displayed un unit stays at around 5V.
Checking with a 18W USB PD powerbank and a 30W USB PD powerbank results in 18W (or 30W respectively) power usage.
Stock firmware (minidso V1.23) is able to negotiate 18W on all devices (I am not sure if 18W may be a limitation of the stock firmware).

* **What is the expected behavior?**
Expected behavior would be that the iron can negotiate full available power from (all) PD capable devices.

***Steps to reproduce the bug:***
Connect iron with power source and read values on display.

* **What is the motivation / use case for changing the behavior?**
Broader compatibility and better usage, smoother user experience and quicker heat up times due to better PD negotiations.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

USB PD wall plug: Anker PowerPort Speed PD30, modell A2014
USB PD powerbank, 18W: SmilePowo YD-690
USB PD powerbank, 30W: PowerAdd Energy Cell II 26800

  - Firmware Version: 2.14.1
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : ratings are as specified by PD and QC3.0 up to max. of rated power of each device

elschopi, 2021-02-25:
I'll close this issue as noone seems interested in it. So I may be the only one affected. I'll just not be able to use my wall adapter I guess.

closed 2021-02-25 by elschopi

Ralim, 2021-02-25:
Hello,

Your not being ignored; it's more that this is a duplicate issue to all of the other PD issues already discussing adapter's that have quirks, and those that have extra debugging information normally get attention.

Most likely when those are resolved your adaptor will magically work.