Ralim/IronOS #841: Pinecil not keeping battery pack awake

davidthejr, 2021-02-03:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
* Using usb battery pack with a Pinecil, but when either holding a temp, or sitting at idle the battery pack with turn off.  Happens at default powerpulse of .5 and max of 4.9.

* **What is the expected behavior?**
* while the iron is on, even in idle it should keep the battery pack awake.

***Steps to reproduce the bug:***
Plug into the qc3 port on the battery pack, turn battery pack on, wait.

* **What is the motivation / use case for changing the behavior?**
* ease of use with a battery pack

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.14.2425902
  - PCB Version: "PCB 3"
  - Power Supply (Voltage and Current Rating) : Tronsmart PBT12, QC3 @ 12v/1.5A

Ralim, 2021-02-06:
Hi,

If your battery bank is not staying awake with the 5W pulse what would you like the firmware to do?
@5W pulses you are already starting to affect soldering and temp control accuracy.

Increasing the duration / repetition rate / power limit will result in loss of temperature control.

You are probably in the unlucky group of people who has a power bank with too aggressive of a shutoff mechanisim.
Have you read all of the other threads around this and tried the other mitigations people have tried?

River-Mochi, 2023-03-24:
@davidthejr   if this issue is resolved  you could  hit the **purple button** at the bottom  to close the ticket 

-  in the Advanced  menu, you can find Power pulse which you can experiment and adjust but  it could cause some issues with how Pinecil functions depending on what you set it to, the default  is 0.5 pulse.
- I suggest updating to the newest firmware first.
- see here https://ralim.github.io/IronOS/Settings/#setting-power-pulse
Setting: Power pulse
Enables and sets the wattage of the power pulse. Power pulse causes the device to briefly turn on the heater to draw power to avoid power banks going to sleep.