Ralim/IronOS #842: Temperature stuck at 463 deg

balrog-kun, 2021-02-04:
_This is a Question_

This is probably a hardware issue, I wonder if anyone has a pointer to which PCB element might need replacing.  I had the mosfet burn out and the tip glow red hot.  Replaced the mosfet and now everytime the TS100 starts heating up it quickly goes to 463°C and waits for the tip to cool down but that never happens.  I suspect the sensor sees a constant temperature and the only reason it shows lower values the first 500ms is due to the PID or low pass filter.

Tried two different tips.  Any ideas what may be wrong?

PCB version probably 1.

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

Ralim, 2021-02-04:
Most likely you have broken the op-amp in the readback path.

Can you dump all the values from the debugging menu and we can confirm all else is as it should be

balrog-kun, 2021-02-04:
Good tip, very appreciated! I just upgraded from v1.08 to latest and this one directly says `NO TIP`, here's the debug menu dump:
```
17-01-21
HW G 197    (206 on USB power)
HW M 189  
HW P 96
Time (works)
Move (works)
RTip 10407  (~830 on USB)
CTip ~415   (30 on USB)
CHan ~289   (~301 on USB)
Vin (works)
PCB   1
PWR  0
Max  413
```

Ralim, 2021-02-04:
Hi,

Yeah that looks to me like something in the readback path is broken.
Suspects are this block:
[image]

balrog-kun, 2021-02-04:
Awesome, I have the list of things to test now.  Closing the issue to avoid inflating the issue list but I'll post an update if I find which element(s) was the culprit, for google's reference.

closed 2021-02-04 by balrog-kun