Ralim/IronOS #843: Software frozen in solder mode with tip heated

nitz, 2021-02-05:
_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "official" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
I was using my iron (Pinecil), and had just cleaned the tip in my wool brush, and went to switch the iron into cooldown mode, and noticed I couldn't switch modes. In fact, none of my input was working, the the screen/temperature wasn't updating, but the iron was definitely still hot.

* **What is the expected behavior?**
Input should continue to be accepted.

In the case of a hard lock like that, I'd expect a WDT to rest the iron at the very least, so it could start cooling down.

***Steps to reproduce the bug:***

- Turn iron on
- Switch to solder mode
- Solder some things
- Clean tip
- Repeat

I want to say it was something to do with the tip cleaning, perhaps it dislodged just a bit when it was trying to read the temperature or something, and got stuck in an infinite loop?

Here's a short video of it behaving this way: [twitter link](https://twitter.com/_nitz/status/1357522998048288768)

* **What is the motivation / use case for changing the behavior?**

The iron not shutting off/being able to shut off is dangerous!

* **What are you running:**

  - Firmware Version: 2.14.2425902
  - PCB Version: 3
  - Power Supply (Voltage and Current Rating) : 15V, 3A [link](https://www.anker.com/products/variant/powerextend-usbc-3-capsule/A9132121)

Ralim, 2021-02-05:
This is something I'm trying to work on a good way to solve.

The issue is the I2C driver in the mcu getting stuck _somewhere_.

I believe something is breaking here: https://github.com/Ralim/IronOS/blob/1f6a3ad16728992d50e8992317ebf83edc8bafab/source/Core/BSP/Pine64/I2C_Wrapper.cpp#L26

The unit does not WDT as that is hooked into the PID and temp controller loop, not the GUI.

nitz, 2021-02-05:
I'm glad it's on your radar! I was just feeling thankful that I didn't set the iron down and wander off like I usually do, relying on the idle cooldown.

I had a similar I2C issue on a client project. Had a sensor that would have issues reporting data consistently so it had to be poked over and over again, and there was a "read twice til both reads match" loop, and every so often, it just *wouldn't* give the data back twice in a row. Nightmares. 😅

Thanks for the quick response!

Ralim, 2021-02-24:
@nitz 
If you get time; would be swell if you could give[ this build](https://github.com/Ralim/IronOS/actions/runs/595440853) a spin and see if it helps with the issues for you. It has for me here, but as its intermittent i may have just been lucky.

nitz, 2021-02-24:
Sure thing! I will flash it this evening!

nitz, 2021-02-26:
Seems good so far! I've poked and prodded it it a bit as well and haven't reproduced the problem yet :)

nitz, 2021-04-15:
I used it a few times since then and haven't reproduced the problem! I'm gonna go ahead and close this for now to get it out of your list! 🙂

Thanks!

closed 2021-04-15 by nitz