Ralim/IronOS #846: Idea for temperature calibration compromise: use body temperature instead of room temperature

vatrat, 2021-02-08:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [X] This is not a bug in the vendor bootloader (aka DFU)
  - [X] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
Partially calibrated temperature using soldering iron's internal temperature sensor, expecting the body to be "cold" (room temperature, unknown).
* **What is the expected behavior?**
I propose that, since these irons are so small, you could set a mode to cool down until it stops cooling down, then grab the tip and the iron with both hands to give a 36.1C-37.2C (avg supposedly 37C) temperature reference.
***Steps to reproduce the bug:***

- make sure iron is off
- leave iron tip and body in room temperature location for enough time for it to cool
- Turn on...
- Do not hold iron (just press buttons)....
- go to settings menu>advanced>temperature calibration
- continue calibration
- iron is sort of calibrated

* **What is the motivation / use case for changing the behavior?**
-Ease of setup/accuracy of calibration
-iron tells you to make sure the iron is completely cool (room temperature)
- body temperature would offer more precise and accurate calibration results. (OR a number between body temperature and "room temperature") 
- body temperature: roughly 37 +/- 0.6 C (some new research claims 36.4 is the new average, I don't see a range for that)
- room temperature: roughly 22.5 +/- 2.5 C (20 - 25 as commonly defined, wider variance in non-home use) 
- for example, "room temperature" for me is usually 23.5C if air conditioning is running. If someone else used 20C in code, 3.5C would be added to my calibration error, in addition to the normal thermal noise, random error, and steady error.
- One challenge for this would be knowing when your iron body's internal temperature sensor has roughly equalized temperature with your body temperature.

- Another way to solve this issue and sync your iron with non-contact temperature sensors would be to allow modification of the rough actual value for room temperature using buttons before the temperature calibration process.
- This calibration temperature expected reading value would allow you to improve calibration if you really wanted to, not that it would be necessary past this point.
- This change would not conflict with a new thermal curve or heating system in software, it only gives more information to the CPU at the time of temperature calibration.

If I get time, I'll try to set this up. 

* **What are you running:**

TIS100, preinstalled bootloader, most recent version (2.14.1) .hex release

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.14.1
  - PCB Version: (1/2) unknown
  - Power Supply (Voltage and Current Rating) : 15.7v, 3.5A(?)

Ralim, 2021-02-08:
The current system is waiting for the handle temp == iron tip temp; not room temperature.

This is done as we want these two to be equal such that the metals do not generate any voltage potential from the temperature difference between the handle and the tip.

I dont see how this would help ultimately with that, unless you intend to hold the tip and iron in your hand until the internal temp sensor == the iron tip. In that case you could just do that with no software changes?

discip, 2022-04-21:
@vatrat
Please consider closing this issue. 😊

vatrat, 2022-04-21:
Sure. Closing, idea didn't make sense

closed 2022-04-21 by vatrat