Ralim/IronOS #848: PD charger/power supply resets when negotiating PD protocol (Pinecil)

fire3element, 2021-02-09:
PD power supply cuts out when the Pinecil negotiates voltages above the initial 5v.
I realize there are already a few issues open regarding this similar issue, but I want to provide some data I have collected which might help identify the root cause.

_This is a [Bug]_

* **I have**
  - [*] Searched previous issues 
  - [*] This is in _this_ firmware
  - [*] This is not a bug in the vendor bootloader (aka DFU)
  - [*] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
* When the Pinecil attempts to negotiate a voltage above the initial 5v, the PD power supply will ramp up to the requested voltage then drop out (or reset). This is an infinite loop. The only way to stabilize the connection is to use an intermediate device between the power supply and the iron to request the desired voltage, then plug in the Pinecil. 

* **What is the expected behavior?**
* Plug in PD power supply > Pinecil requests maximum supported voltage > Request completed and power supply stabilizes

***Steps to reproduce the bug:***
- Plug in PD power supply to Pinecil
- Observe PD power supply attempting to reach requested voltage no the Pinecil's screen
- Pinecil resets and cycle begins again

* **What are you running:**
  - Firmware Version: v2.13.286119A
  - PCB Version: 3
  - Power Supply (Voltage and Current Rating) : Brand: Targus | M# APA93US | Voltage: 5v3A, 12v3A, 15v2.5A, 20v2.25A | PD ver: (I think it is 2.0)

I have a USB diagnostic meter called the Witrn U2P. It has the ability to monitor voltage, amperage, requested protocols, data line voltages, and even raw data from connected PD devices. I have used this tool to gather some possibly helpful and insightful info.

Below are a list and description of all files provided for informative and diagnostic value:
[image]
DSC_3076.jpg - image of data plate on the Targus PD power supply.

https://user-images.githubusercontent.com/44314623/107403520-d1c1a900-6aca-11eb-93bd-4f3b86aac8fa.mp4
MOV_3077.mp4 - In this instance, the PD supply is already plugged into the USB meter and supplying 5v. The Pinecil is then plugged into the meter and the D+ and D- jump to 3.3v. After a few moments, it appears the Pinecil is attempting to negotiate a higher voltage, but fails. You can watch the D+/D- data lines and the protocol detected alternate. Eventually the Pinecil stops sending the request.

https://user-images.githubusercontent.com/44314623/107403728-164d4480-6acb-11eb-83b9-8932c079358e.mp4
MOV_3078.mp4 - In this instance, the Pinecil is already connected to the USB meter, then the PD supply is plugged in. You can see on the Pinecil display the voltage ramping up to 15v, then the PD supply resets. This is the behavior observed when you plug in the PD power supply directly to the Pinecil.

https://user-images.githubusercontent.com/44314623/107403776-26652400-6acb-11eb-8c4c-f0d0d17c0981.mp4
MOV_3079.mp4 - In this instance, I am powering the USB meter externally with a different power source. This is done to observe the protocol request behavior and activity on the data lines. This is the same condition as explained in MOV_3078.

https://user-images.githubusercontent.com/44314623/107403828-367d0380-6acb-11eb-9b99-97140bfa9be7.mp4
MOV_3080.mp4 - This is the same condition explained in MOV_3078. I am using the USB meters oscilioscope like graphing function to show the voltage request, stepping, and reset characteristics.

https://user-images.githubusercontent.com/44314623/107403921-4c8ac400-6acb-11eb-81df-88a3cf9e6817.mp4
MOV_3081.mp4 - In this instance, I am using the USB meter's raw data capture function. All the captured data from the communication between the PD supply and the Pinecil is displayed here.

[image]
Manual.png - Explanation of the captured data from the USB meter product manual.

Gwebber-Net, 2021-02-09:
1. Can you update to latest firmware ?
2. How certain are you its not the Power Supply that is uncapable ?

DooMMasteR, 2021-02-11:
This is the same behavior my Anker PowerPort+ 5.
USB PD Output: 5V=3A/9V=3A, 15V=2A/20V=1.5A
See #838 which might be the same issue.

Ralim, 2021-02-24:
That raw data capture function is fantastic for this. (though the reverse numbering is a tad odd).

You can Ignore usb lines toggling, thats QC starting after PD fails.

So the PD stack will silently ignore vendor specific messages; or messages that it does not support that do not require an explicit response.

There are also a bunch of messages we NACK:

* Direction swap
* Source Capabilies (as we cant source)
* PR swap
* VConn swap
* Requests
* Sink capabilities
* Go To Min / Giveback
* Battery status

Looking at the log however, it looks like its not us NACK'ing or sending a softreset as those should show on the message capture.

What would be really nice is to truncate that log to show one "cycle" i.e. plug in until it resets, then disconnect pinecil. 
Just to confirm that its not the power supply sending multiple SRC CAP messages thats breaking things.
(Its normal for PSU to send a few in a row though).

Ralim, 2021-02-24:
Additionally, I just swapped this branch to ignore unknown rather than soft reset.
Could you give it a trial for me when you have time :)
https://github.com/Ralim/IronOS/actions/runs/595467155

fire3element, 2021-02-25:
New video showing one power cycle.
https://user-images.githubusercontent.com/44314623/109108989-2d5e7a00-76fa-11eb-97d9-a659c4c59250.mp4

I flashed and tested the update you requested. Behavior remained consistent with previous firmware.
The captured PD data also remained identical to that shown in the video.
( for confirmation, updated firmware on Pinecil shows v2.14.A01E32D )

If you require further testing or data, let me know.

Ralim, 2021-02-25:
Hmm interesting and thank you for the record.

Do you per chance have something else you can connect that uses >5V and record to compare?

Good that it doesnt change anything (as it shouldnt). Frustrating as that means that the firmware is "correct" in that its just not answering the vendor messages and is silently ignoring them. I would be curious to see what other devices do :/

fire3element, 2021-02-26:
I have tested the Targus PD power adapter (M# APA93US) with the Nintendo Switch (undocked) and an HP 2in1 notebook/tablet.
I also tested the original Nintendo Switch PD power supply (M# HAC-002) with the Pinecil, Switch (undocked), and HP.
The HAC-002 powers the Pinecil fine at it's maximum output voltage of 15v, which is why I have included it in the tests.

APA93US > Switch
https://user-images.githubusercontent.com/44314623/109270545-c9a87f80-77d3-11eb-9335-03a8eb00c23e.mp4

APA93US > HP
https://user-images.githubusercontent.com/44314623/109270606-e349c700-77d3-11eb-94de-fe2c222e0c57.mp4

HAC-002 > Pinecil    (no odd behavior observed. Pinecil powers up and ramps to 15v without reset)
https://user-images.githubusercontent.com/44314623/109270634-f5c40080-77d3-11eb-9c00-992dcb00d854.mp4

HAC-002 > Switch
https://user-images.githubusercontent.com/44314623/109270822-39b70580-77d4-11eb-92de-98aec4086133.mp4

HAC-002 > HP
https://user-images.githubusercontent.com/44314623/109270792-27d56280-77d4-11eb-96cd-00c15ea4f182.mp4

Ralim, 2021-03-04:
These are very interesting!!
Apologies on delay as I was testing some of this locally here too.
I believe the issue is the DR swap and the DR swap rejection is not playing out how it should for this firmware.
I have managed to find one dock that sends a DR swap and then the firmware rejects it (as expected) and another where this does not occur.

Trying to figure out _why_ it does not get rejected... Is not obvious to me yet but this is interesting as it looks like the same thing might be occuring with the APA93US

fire3element, 2021-04-28:
@Ralim  Apologies it took me so long to respond. Had issues getting DFU mode to work, and ended up shelving it for a while.
Finally got around to messing with it again tonight. Had to manually flash in Linux, as Windows was being uncooperative.

I can confirm from my end that the beta build fixed the PD cycling issue on my hardware. Excellent work sir.
Thank you for your dedication and resolution to this issue.

Ralim, 2021-04-28:
@fire3element No issue at all on the delay; we are all busy people.
Glad that works for you !

fire3element, 2021-04-28:
I'll close the issue now as the problem has been resolved on my end. Thanks again.

closed 2021-04-28 by fire3element