Ralim/IronOS #853: Pinecil custom bootup logo not showing up

loveless-u, 2021-02-16:
_This is a [Bug/Question]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
The device goes straight to menu after power-on.
* **What is the expected behavior?**
The device shows my cool logo after power-on.
***Steps to reproduce the bug:***
-Flash latest release (without MASS_ERASE, as that does not work, in accordance with #820)
-Flash image according to docs (tried a presupplied one too for sanity)
-Power on

  - 2nd batch Pinecil
  - Firmware Version: 2.14.2425902
  - PCB Version: 3
  
  During flash, dfu-util 0.10 complains:  `Found GD32VF103, which reports a bad page size and count for its internal memory.`, but proceeds to flash.

  
  
  This might be related to #820

alvinhochun, 2021-02-16:
@Ralim can correct me if I am wrong, but I believe the issue here is that the documentation on the master branch has been updated to reflect the change of logo address made in #830, which now expects the logo to be at `0x0801F800` for the Pinecil. The latest release was made before that, which expected the logo to be at `0x0800F800`.

The solution is to either flash the logo to `0x0800F800`, or flash the latest CI build: https://github.com/Ralim/IronOS/actions/runs/568576665 (scroll down to under "Artifacts").

The dfu-util message is normal and is irrelevant.

loveless-u, 2021-02-16:
@alvinhochun Changing the offset did work, thanks!
Maybe the documentation for the current release should stay visible somewhere, in case things like these happen?

closed 2021-02-16 by loveless-u