Ralim/IronOS #854: Grounding/earthing, ESD and C26

SuperNeko, 2021-02-18:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Question]_

* **I have**
  - [x] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

I have a question regards grounding which bothered me for sometime. 

TS100 has very uncomfortable grounding connection - it requires addition wiring. 
So I researched for solution and found out that it is possible to remove C26 and short pads. That will short tip and DC negative. 
My power supply's DC negative is grounded, so that way my tip is now grounded to mains ground. Using it for a while proved that it had no negative consequences.

_So, long story short: from ESD perspective, if soldering iron is not grounded via grounding screw, but power supply's DC negative is grounded, will C26 discharge static from tip?_

Because I think maybe it was not necessary to short tip to ground to have comfortable and ESD safe setup.

Sorry, if it is stupid question, I am just a beginner in electrical stuff.

alvinhochun, 2021-02-19:
This is just my rookie, non-professional view:

What you usually want is to keep your body, the iron tip, the workbench, the components and the board you are working on at the same electrical potential and prevent static build-up on any of them. The common way AFAIK is to connect them to a common ESD earthing point with high impedance paths (at least 1M ohm), and then connect it to mains earth.

If you rely on earthing the iron via the iron's power supply GND, then the iron will not be using the common ESD earthing point for the rest of your setup, so it may not be as effective, Also, if the power supply connects the secondary GND to mains earth directly without a resistor in between, there is a risk of electric shock or short-circuit when working with live equipments (the RCD and MCB will probably save you, and you really shouldn't be soldering on live stuff, but still). Practically it should still work fine for most hobbyist uses.

I would like to hear some comments from experts too.

SuperNeko, 2021-02-28:
I edited my post, because I put question wrong - I was more interested in C26 purpose rather than ESD stuff. Sorry for confusion.

Ralim, 2021-05-01:
I believe the purpose of C26 is to loosely ac couple the tip to the power supply negative for emi reasons.
Same as why you have the cap from output GND to earth on an ac adapter.

You should be totally fine to swap it out. I would probably replace it with like a 1M resistor.

closed 2021-07-04 by SuperNeko