Ralim/IronOS #855: Request for adapters table on doc

Essetekip, 2021-02-22:
Hello, in the documentation or other, would it be possible to make a table with the tested and recommended power supplies? Especially for USB PD adapters and cables, the most used at least

discip, 2021-05-04:
What about:
## **[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**
Works great so far!

River-Mochi, 2023-03-23:
the Pinecil Wiki  has an entire article dedicated to different power supplies 
**https://wiki.pine64.org/wiki/Pinecil_Power_Supplies**

however  each iron uses different  chargers.  for example the  EPR PD3.1, 28V-140w chargers, only the Pinecil V2 can use the 28V  on that usb-C charger.   other  Irons, people would have to drop to a simple PD60 or PD100 cable to use that same  class of 28V usb-C charger. 

@Essetekip   perhaps you could make a document in the IronOS  wiki pages  just for the specific irons  that you are testing chargers on and list?  There is also a Discussions topic on  chargers where people add in chargers they use for different irons but as each iron has different Specs, it would have to be separated by  Iron brand/model.

For Pinecil from Pine64,  there is a dedicated wiki that is robust and detailed and no point copying the  whole thing over to here.  most people with Pinecils at least just go to Pinecil Wiki pages.   That one article is just onthe Power supplies, there is another complete article just on[ Pinecil  Cartridge tips ](https://wiki.pine64.org/wiki/Pinecil_Tips)
Other  issue is sometimes links to chargers go bad, meaning they are no longer sold. That is why the Pinecil Wiki pages also  gives  specific  ratings people should look for in buying a good performance charger rather than just links alone.

As this is IronOS  firmware,  and not the hardware side,  the Pinecil Wiki is kept separate.

River-Mochi, 2023-03-23:
I'm not sure this is an  Issue Ticket and perhaps more Discussion. 

@Essetekip   you might want to close this ticket or transfer it over to discussions which may be more appropriate.