Ralim/IronOS #859: Broken wiki links

adatum, 2021-03-05:
Clicking on the "Installation/Flashing" or "Mods" links in the wiki sidebar redirects to the "Home" wiki page, with a GitHub banner message saying "You do not have permission to update this wiki."

At least the installation/flashing information is available at https://github.com/Ralim/IronOS/blob/master/Documentation/Flashing.md

Ralim, 2021-03-07:
Removed the dead links.

Wiki is locked to prevent people adding in commision links to parts again. _sigh_

closed 2021-03-07 by Ralim

adatum, 2021-03-07:
Is there a reason why the wiki does not include the documentation? I thought the link was meant to point to the documentation.

Ralim, 2021-03-07:
Documentation is stored in the repo rather than the wiki so that it is kept up to date with the firmware.