Ralim/IronOS #860: Broken uppercase cyrillic char "З"

playday3008, 2021-03-06:
_This is a [Bug]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
1. РОзБЛОКОВАНО
2. змінювати температуру при короткому натисканні!
3. змінювати температуру при довгому натисканні!
* **What is the expected behavior?**
1. РОЗБЛОКОВАНО
2. Змінювати температуру при короткому натисканні!
3. Змінювати температуру при довгому натисканні!

***Steps to reproduce the bug:***

- Install TS100_UK.hex
- Try to block screen then unlock
- Or check TempChangeShortStep/TempChangeLongStep description

* **What are you running:**

  - Firmware Version: last at this moment
  - PCB Version: 1

playday3008, 2021-03-07:
And what is this? 
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L13
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L15
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L34
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L42
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L47
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L70
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L73
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L104
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L216
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L220
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L523
The same is written twice

playday3008, 2021-03-07:
Also better change this
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L690
to this
`0x7C,0x40,0x7C,0x40,0x7C,0x00`

playday3008, 2021-03-08:
@Ralim Check [this](https://github.com/playday3008/IronOS/tree/fix-chars), I added `З` char to `fontTables.py`, and... nothing changed. Why?

P.S. Also added Ukrainian `Ґ` and `ґ` chars, and fix `Ш`, `Щ`, `ш` and `щ` in small font map

Ralim, 2021-03-08:
You may need to check that the symbols are being decided by python correctly and it's not mangling what code point you want them to actually be at;l.

When it generates the font table is it using the font data from a different location ?

playday3008, 2021-03-08:
i checked generated cpp translation file, and
Translation_UK.cpp:
```cpp
0x00,0x60,0x70,0x30,0x30,0x30,0x30,0x30,0x30,0xF0,0xE0,0x00,0x00,0x18,0x38,0x30,0x33,0x33,0x33,0x33,0x33,0x3F,0x1D,0x00,//\x3B -> З
```
[image]

playday3008, 2021-03-08:
Why it replace uppercase char to lowercase

Ralim, 2021-03-08:
My assumption is this is something to do with Python. Honestly no real idea myself though, I haven't run into it before

playday3008, 2021-03-08:
Maybe he can help, he wrote that script @t-8ch

t-8ch, 2021-03-08:
@playday3008 I didn't write that script in particular.
But my Makefile refactoring did not track the dependency to `fontTables.py`.
As a workaround you can touch `make_translations.py` and run `make` again to rebuild the translations tables.

t-8ch, 2021-03-08:
Looks to be this line:
https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L253
It defines the uppercase character to the lowercase glyph.

playday3008, 2021-03-08:
wow, 

> Looks to be this line:
> https://github.com/Ralim/IronOS/blob/955b7346d41096a6a88c1d6a411ab346bdb9e6d3/Translations/fontTables.py#L253
> 
> 
> It defines the uppercase character to the lowercase glyph.

Wow, I didn't even notice it

PR #865 (2021-03-08): Fix Cyrillic glyphs
* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**
(Bug fix.)

Fix #860
closed 2021-03-08 by Ralim