Ralim/IronOS #869: Seamless settings upgrade instead of resetting

alvinhochun, 2021-03-11:
_This is a Feature Request_

* **I have**
  - Searched previous issues 
  - This is in _this_ firmware, not vendor "offical" firmware
  - This is not a bug in the vendor bootloader (aka DFU)
  - I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

Whenever the firmware bumps the `SETTINGSVERSION`, the settings have to be reset.

* **What is the expected behavior?**

The firmware should be able to upgrade the settings from at least the last previous release version whenever compatible.

* **What is the motivation / use case for changing the behavior?**

It is annoying to have the settings reset on upgrade, especially for the input voltage and temperature calibration settings, since one cannot simply reapply the exact values manually from the menu.

playday3008, 2021-03-12:
I guess we needs to store every version of settings struct in firmware file to compare it and save changes, probably we can't do it because it will increase the size of the firmware

Ralim, 2021-03-12:
We could solve this by having a migration function + making the settings struct append only mostly.
It is extra work that I'm happy to take a PR on, but dont have time right now to implement :/

alvinhochun, 2021-03-14:
> 
> 
> I guess we need to store every version of settings struct in firmware file to compare it and save changes, probably we can't do it because it will increase the size of the firmware

So I said "the last previous release version whenever compatible".

> 
> 
> We could solve this by having a migration function + making the settings struct append only mostly.
> It is extra work that I'm happy to take a PR on, but dont have time right now to implement :/

Unfortunately, the last release 2.14.1 (`SETTINGSVERSION == 0x24`), the `temperatureInF` setting definition is conditionally compiled in, depending on the build language, which means it is impractical to migrate from this release or any versions before it.

On current master this isn't a problem anymore so we can probably start doing settings upgrade in the release after the next one.

t-8ch, 2021-03-14:
One way to have a wide range of backward compatibility would be to use the length of the settings as the version and only append fields.
This way newer firmwares can load as many bytes as the old firmware wrote and fill the rest with default values.
This also gives forwards compatibility for older firmwares loading newer settings, as they can just discard any data beyond their longest known settings structure.

(I think I have read about this being used to handle extensible syscall argument structures on Linux)

Notes:
* A maximum of 256 bytes of settings can be represented without breaking compatability
* The current settingsversion is 38 (0x24) but the size of `systemSettings` is only 32 bytes, so this needs special handling

Ralim, 2021-03-14:
I'm not against doing append only.

Only thoughts are:
* Will bitfields always pack the same
* Ram consumption over time

federicodilo, 2021-03-14:
Is a "firmware builder" a bad idea?

Ralim, 2021-03-15:
I basically agree with both of your pro's and cons.
I'm not against one that just merges firmware with a pre-setup settings / logo into a single flashable hex file though.

Also dont want it to be a mega burden to support as a dev

_hmm_

alvinhochun, 2021-05-08:
Now that we have a "stable" checkpoint (v2.15-rc1), shall we start to commit to this?

> One way to have a wide range of backward compatibility would be to use the length of the settings as the version and only append fields.
> This way newer firmwares can load as many bytes as the old [...]

While this is one way to do it, I think this is not necessarily a good idea because:

- It forbids using bitfields for new settings (not that it is a bad thing though).
- Future firmwares may remove settings, which leaves behind garbage values and may cause issues when downgrading, requiring a reset anyway.

I cannot deny that allowing for downgrading is a nice benefit, but to most users it is perhaps not very relevant. I assume most of the time downgrading is done to test for regressions, and it is likely that a reset will be needed for proper testing. The actual benefit of this is perhaps close to null.

So I think a one-way upgrade is good enough.

> Will bitfields always pack the same

As far as I understand, bitfield allocation is **implementation-defined**, so as long as we use GCC the bitfields should remain compatible. The behaviour of GCC is to allocate bitfields starting from the LSB first.

---
My thinking is to simply make modifications to the struct mostly compatible, i.e. append when adding new fields. Deleted settings fields will become padding. If it is possible to insert new fields or bitfields to the existing padding space without affecting the rest of the fields, then it is also fine to do so (but perhaps not really a good idea).

When upgrading the settings, it just needs to reinterpret the bytes and set the new fields to the default value and change the version field.

The thing is, I would really like to have tests to verify that this upgrading works properly. I don't think we can do type punning with either `union` or `reinterpret_cast` in `constexpr` context, so we can't just verify it using compile-time `static_assert`, which means it will have to be either:

- cross-compiled tests that can be run via QEMU in some way, or
- a test firmware that one needs to manually flash to the Pinecil/TS100/whatever to verify, and done once per model.

The former I have no idea how to do, and the latter seems too troublesome. Could there be another way?

discip, 2022-02-12:
@alvinhochun & @Ralim
This is implemented already!

Can be closed.