Ralim/IronOS #871: Submenus, text scrolling/animation speed

Sinclair-ZX81, 2021-03-13:
_This is a Complaint_

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [X] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
When diving into submenus or leaving them, the OLED text scrolling behaviour is way too fast / almost not perceivable

* **What is the expected behavior?**
The perfect animation speed was with FW v2.10.0

* **Steps to reproduce the bug:***
Go into any submenu or leave it and watch the scrolling speed of the text

* **What is the motivation / use case for changing the behavior?**
The slower scrolling speed behaviour of v2.10.0 was looking much better and did help navigating through the menus, especially when you haven't memorized all the menus

* **What are you running:**
Firmware Version: 2.14.1
PCB Version: 1
Power Supply (Voltage and Current Rating) : 20.4 VDC

@Ralim - please can you bring back the scrolling / animation speed as it was with v2.10.0 ?

Ralim, 2021-03-13:
Hi,
This is a _Bug report_ not a _complaint_.

But yes I am having a look now.

Ralim, 2021-03-13:
Please test the appropriate file from here to check its resolved for you: https://github.com/Ralim/IronOS/actions/runs/650046462

Sinclair-ZX81, 2021-03-14:
Thanks for your quick response! [image]

Build 2.15.C7784FD does it fix in each of my two TS100 (PCB: 1)

Please don't forget to adjust timers (?) for the info text scrolling, as the fix has slowed it down very much, see recording.
[image]

https://user-images.githubusercontent.com/80591481/111069962-cc8fb980-84cf-11eb-93e5-66179c8c0f9f.mp4

PR #874 (2021-03-15): Correct a few missed time values from the tick rate change
This should resolve #871 I believe.

A while the tick rate of the FreeRToS kernel was changed 1kHz mostly to keep Pinecil and Miniware the same.
Obviously some of these were missed during that conversion, hopefuly this is most of them tagged correctly now.
closed 2021-03-15 by Ralim