Ralim/IronOS #881: some bugs in latest commit: be91f78 (scrolling does not stop | misaligned values)

discip, 2021-03-15:
_This is a [Bug Report]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "official" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked, this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
The previous behavior of pausing at the highest value, when holding the **A- button**, is gone.

* **What is the expected behavior?**
When the last value is reached, the ascension stops, while the scroll 'bar' is blinking.
And starts from the beginning, only if button **A** is pressed again.

* **Is the issue reproducible.**
Yes

  * ***Steps to reproduce:***
    1. Go to any entry where a value can be changed.
    2. Press and hold button **A**.
    3. Wait for it to reach the highest value.
    3. Ask yourself: *'Hey... , shouldn't this has stopped at the highest value?'*.

* **What is the motivation / use case for changing the behavior?**
To help, making the iron function as it was intended to.

* **Additional information**
   ### The value of newly implemented entry: `Anim. speed` is misaligned. The values should be aligned on the right side.

* **What are you running:**
  topic|**TS80P**
  -|-
  Firmware Version|**https://github.com/Ralim/IronOS/commit/be91f7833d3c74e3ca1fecc3387eadf8a49d69a5**
  PCB Version|**2**

Thank you in advance
kind regards

devydd, 2021-03-16:
Both issues (not stopping on the last item & misaligned value for Animation speed) confirmed on TS80 from master https://github.com/Ralim/IronOS/commit/be91f7833d3c74e3ca1fecc3387eadf8a49d69a5

This PR fixes the issue with no hold time: https://github.com/Ralim/IronOS/pull/882

devydd, 2021-03-16:
And this PR fixes the misalignment issue: https://github.com/Ralim/IronOS/pull/883

closed 2021-03-16 by Ralim