Ralim/IronOS #886: dash instead of values in 'Power pulse wait time' and in 'Power pulse duration'

discip, 2021-03-16:
_This is a [Bug Report]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "official" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked, this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
The only symbol shown as value of the newly implemented entries: **`Power pulse wait time`** and **`Power pulse duration`** is a ***dash***.

* **What is the expected behavior?**
The respective value should be represented, where now there is the ***dash***.

* **Is the issue reproducible.**
Yes

  * ***Steps to reproduce:***
    1. Go to **Advanced options**.
    2. Now go to **`Power pulse wait time`** or **`Power pulse duration`**.
    3. Now you see the mentioned ***dash***.

* **What is the motivation / use case for changing the behavior?**
To help, making the iron function as it was intended to.

* **What are you running:**
  topic|**TS80P**
  -|-
  Firmware Version|**https://github.com/Ralim/IronOS/commit/ecf4651f6df5004fef1031cd23e54a385ff1ae24**
  PCB Version|**2**

Thank you in advance
kind regards

Ralim, 2021-03-16:
Hmm, you should be able to adjust it from there; looks like dash is the "off" state.

I'll have a look when I can tonight though

discip, 2021-03-16:
@Ralim
Good evening sir,
no it is not the *'off state'*, since the value remains a ***dash*** no matter how often the **A button** is pressed, however it seems to increase in the background, because at some point the 'scroll bar' starts to blink (For both entries the same!).

Ralim, 2021-03-17:
I think what may have been occuring is that you were attempting to adjust the duration with the pulse rate turned to off.
I _have_ however found a bug that prevented me turning it off 😂  that ill push a fix for shortly

discip, 2021-03-17:
@Ralim
You are totally right about that! 🤦‍♂️
Nevertheless, I think, it would be much more convenient, to make these entries disappear, if  **`Power pulse`** is turned ***off***.
regards

Ralim, 2021-03-17:
I agree.
Shall fix that tonight (reopening ticket)

devydd, 2021-03-17:
> I think what may have been occuring is that you were attempting to adjust the duration with the pulse power turned to off.

Yes, that was my intent. You have to enable power pulse before changing settings related to the power pulse.

And I agree I would be better to remove those two options altogether from the menu when the power pulse is off.

Ralim, 2021-03-18:
@discip Can you give latest master a spin?
I _think_ that should handle it :)

discip, 2021-03-18:
@Ralim
That worked! 😃👍
Thank you!

closed 2021-03-18 by discip

devydd, 2021-03-18:
Working as expected. Confirmed on TS80.