Ralim/IronOS #888: I use command building release, it didn't work.

laivn, 2021-03-17:
_This is a [Question]_

* **I have**
  - [ ] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
I use command building release, it didn't work. 
* **What is the expected behavior?**

***Steps to reproduce the bug:***

- Turn on...
- Hold....

Building firmware for TS100 in BG
make: arm-none-eabi-as: No such file or directory
make: arm-none-eabi-gcc: No such file or directory
make: arm-none-eabi-gcc: No such file or directory
make: arm-none-eabi-gcc: No such file or directory
make: *** [Objects/TS100/./Core/BSP/Miniware/port.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Objects/TS100/./Startup/startup_stm32f103t8ux.o] Error 1
make: *** [Objects/TS100/./Core/BSP/Miniware/BSP_PD.o] Error 1
make: *** [Objects/TS100/./Core/BSP/Miniware/system_stm32f1xx.o] Error 1
make: arm-none-eabi-gcc: No such file or directory
make: arm-none-eabi-gcc: No such file or directory
make: arm-none-eabi-gcc: No such file or directory
make: *** [Objects/TS100/./Core/BSP/Miniware/Setup.o] Error 1
make: arm-none-eabi-gcc: No such file or directory
make: *** [Objects/TS100/./Core/BSP/Miniware/stm32f1xx_it.o] Error 1
make: *** [Objects/TS100/./Core/Src/FreeRTOSHooks.o] Error 1
make: *** [Objects/TS100/./Core/BSP/Miniware/flash.o] Error 1
make: arm-none-eabi-gcc: No such file or directory
make: arm-none-eabi-gcc: No such file or directory
make: *** [Objects/TS100/./Core/Src/freertos.o] Error 1
make: *** [Objects/TS100/./Core/Src/syscalls.o] Error 1

Ralim, 2021-03-17:
Hi,
That would indicate you have not installed the arm compilers on your system before running the command.

laivn, 2021-03-17:
Thx, I settled.

closed 2021-03-17 by Ralim