Ralim/IronOS #892: make menu even clearer by using more animations

discip, 2021-03-18:
@Ralim
_This is a [Feature Request]_

* **What is the current behavior?**
1. If you press, in **`idle screen`**, **B**, the screen slides to the ***left***.
2. If you keep pressing **B** the screen *just* ***changes*** to the next **item (caption)**.
3. If you however enter an **item** by pressing **A** the screen shifts again to the left and by pressing **B** you *'move'* through the respective *'submenu'*, which is indicated by the screen ***changing*** to the next entry and the downward moving scrollbar.
4. If the end of a submenu is reached, the next **caption** ***appears the *same way* the submenu entries would do***.
5. If the end of the main menu is reached, the screen *just* ***changes*** to the **`idle screen`**.

* **What is the expected behavior?**
1. If you press, in **`idle screen`**, **B**, the screen slides ***up***.
2. If you keep pressing **B** the screen keeps ***sliding up*** to the next **item (caption)**.
3. If you however enter an **item** by pressing **A** the screen shifts to the left and by pressing **B** you *'move'* through the respective *'submenu'*, which is indicated by the screen ***sliding to the left*** and the downward moving scrollbar.
4. If the end of a submenu is reached, the next **caption** slides in from the left [ the current screen ***slides to the right*** ].
5. If the end of the main menu is reached, the **`idle screen`** slides in from above [ the current screen ***slides down*** ].

*The enumeration is just to make comparing >before< and >after< easier.* 😃

* **What is the motivation / use case for changing the behavior?**
Guess what!

thank you in advance
kind regards

alvinhochun, 2021-05-03:
> 1. If you press, in **`idle screen`**, **B**, the screen slides _**up**_.

The current behaviour of sliding to the left seems fine to me. I don't really see a reason to change this horizontal transition to a vertical transition.

> 4. If the end of a submenu is reached, the next **caption** slides in from the left [ the current screen _**slides to the right**_ ].

It looks like this was implemented in the past but removed in https://github.com/Ralim/IronOS/commit/04d72cbcd8b6a32b95e4f95902cd040bf82cf35a#diff-669bf661aa47cf2057ddea36ba835fc8623cf175b8e9bba9498d83ccf0e88badL1172-L1184. I can restore this animation, but I'll first need to ask @Ralim if it was removed for any specific reasons.

> 5. If the end of the main menu is reached, the **`idle screen`** slides in from above [ the current screen _**slides down**_ ].

I think it can use a slide-to-the-right horizontal transition. instead, similar to the first item.

discip, 2021-05-03:
@alvinhochun
Thank you for locking into this.
I wanted to have the moving direction separated, to have a clear indication of whether you are in the main menu (vertical) or in a sub menu (horizontal).
Hope this makes sense.
kind regards

alvinhochun, 2021-05-03:
> I wanted to have the moving direction separated, to have a clear indication of whether you are in the main menu (vertical) or in a sub menu (horizontal).
> If everything moves only horizontally, there is not much of such indication.

Well, I'm not so sure about that.

How about you give this build a try? https://github.com/alvinhochun/IronOS/actions/runs/806592344

discip, 2021-05-03:
I will try that, when I am at home later.

discip, 2021-05-03:
@alvinhochun
# Sir ... this is perfect! 👍 Thank you a lot.
I am impressed by the speed, you implemented this with. 🤯
Are you going to implement this into the master built, if @Ralim is good with that?
thank you again
regards

Ralim, 2021-05-04:
> It looks like this was implemented in the past but removed in [04d72cb#diff-669bf661aa47cf2057ddea36ba835fc8623cf175b8e9bba9498d83ccf0e88badL1172-L1184](https://github.com/Ralim/IronOS/commit/04d72cbcd8b6a32b95e4f95902cd040bf82cf35a#diff-669bf661aa47cf2057ddea36ba835fc8623cf175b8e9bba9498d83ccf0e88badL1172-L1184). I can restore this animation, but I'll first need to ask @Ralim if it was removed for any specific reasons.

Honestly cant remember exactly why it was removed. Happy for the animation to be here or not :)

alvinhochun, 2021-05-04:
> Honestly cant remember exactly why it was removed. Happy for the animation to be here or not :)

I see some glitches from the menu skipping code that needs to be ironed out.

> While you are at it, would you mind to change the animation style oft the _**wrench**_, used for the **`Advanced options`** menu, as well?
> I thought about the movement of the real thing _(rotating around one end)_, not sliding in. 😁

IMO better handle that in another issue.

Ralim, 2021-05-04:
>  I thought about the movement of the real thing (rotating around one end), not sliding in. 😁
I would create a seperate issue to track that one

> I see some glitches from the menu skipping code that needs to be ironed out.
Honestly not surprised if there are quirks in there sadly

discip, 2021-05-04:
@alvinhochun 
> I see some glitches from the menu skipping code that needs to be ironed out.

I surely did not test every single feature, but I had no issues (glitches) what so ever.
But if you are saying there are any, I would be really thankful, if you would do the mentioned ironing. 😁

> IMO better handle that in another issue.

@Ralim 
> I would create a seperate issue to track that one

Since this is not crucial, I thought this would not need another issue, but if you say so, I will do.

# Thanks to both of you for this beautiful software, which is definitely supercharging the supported hardware! 😃

kind regards

discip, 2021-05-04:
Closing this,
since @alvinhochun is on the way, to get this done, with this https://github.com/Ralim/IronOS/pull/955#issue-629692915.

# Sir ... thank you a ton! 😁👍

PR #955 (2021-05-05): Impl. menu exit transition
* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

Feature: (re-)add menu exit transition, and completing all menu transitions.

* **What is the current behavior?**

When exiting a menu or submenu, the screen changes directly with no transition animation.

During menu transitions, the menu icon on the main menu isn't drawn.

* **What is the new behavior (if this is a feature change)?**

When exiting a menu or submenu, the screen cha [... cut]
closed 2021-05-04 by discip