Ralim/IronOS #895: Faulty ts100

meelkmeelk, 2021-03-19:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a Faulty problem

* **I have**
  - [✅] Searched previous issues 
  - [✅] This is in original firmware, not vendor "offical" firmware
  - [✅] This is not a bug in the vendor bootloader (aka DFU)
  - [✅] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
1. When turned on, it will show the normal “press” prompt, but after pressing it goes into temperature increasing mode for less than a second and change back to the prompt
2. When plugged into a laptop, it shows “config” on the OLED and any other presses will not do anything except a long press on button A will show the logo
* **What is the expected behavior?**
1. Increase in temperature to default 300 degrees celcius
2. Ts100 should change to DFU mode upon pressing the A button
***Steps to reproduce the bug:***

- Off and on power supply

* **What is the motivation / use case for changing the behavior?**
Startup problem

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.18
  - PCB Version: stm32
  - Power Supply (Voltage and Current Rating) : Input: AC100-240V Output: 12V=3A

Ralim, 2021-03-19:
Hello, if when it connects to a computer it is showing config on the screen then you are not running the firmware from this repository.
You are running the vendor (miniware) firmware.

For support with their firmware please use their forum and email.

However from the sound of it, most likely your power supply is insufficient.

meelkmeelk, 2021-03-20:
Hello, is their a way i could change the firmware? Thank you so much for helping👍

Ralim, 2021-03-20:
Please see the documentation in this repository, linked in the readme.

meelkmeelk, 2021-03-20:
> Please see the documentation in this repository, linked in the readme.

Thank you so much

closed 2021-03-24 by meelkmeelk