Ralim/IronOS #907: Hot swapped tip starts heating up without user input. Risk of burn.

GreekGeorge, 2021-03-23:
_This is a bug that has the risk of burning the user

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [X] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
If a tip is hot swapped, the new tip inserted heats up immediately. I'm not sure if this is normal. Got a slight burn on my fingertips while rotating the tip to position accordingly.

* **What is the expected behavior?**
Press Button A (closer to tip) to start heating up the newly inserted tip.

***Steps to reproduce the bug:***
- Insert tip, bring to heating temperature
- Remove tip
- Insert new tip
- New tip starts heating up immediately

* **What is the motivation / use case for changing the behavior?**
Safety concern. People can burn themselves.

* **What are you running:**

  - Firmware Version: 2.14.1
  - PCB Version: 1
  - Power Supply (Voltage and Current Rating) : 19v at 65W

alvinhochun, 2021-03-23:
It sounds like you accidentally hit the button to enter soldering mode when swapping the tip. It seem unintuitive but it is actually what the current firmware does.

The code can be changed to prevent entering soldering mode when it detects the tip isn't inserted. Perhaps it should even automatically exit soldering mode when it detects that the tip has been removed. However, since the tip removal detection is based on a high temperature reading, it may interfere with operating in a high temperature (say > 410 °C).

GreekGeorge, 2021-03-29:
The way I see it, when there is no tip connected, the iron goes to the menu. But if I'm in the menu and I add a tip, it shouldn't automatically start heating up.
Same thing when hotswapping tips. I guess I can go back to the menu, swap tips, and then heat it up...
idk... It's a bit weird. On one hand it makes sense that hotswapping should heat up the tip asap, on the other hand it's a bit risky.

discip, 2023-05-04:
@Ralim
As @alvinhochun pointed out, detecting a disconnected tip can fail due to the current method.

But what if the rate of temperature rise was used to determine this, since the temperature reading jumps to the maximum value almost immediately when the tip is removed?

e.g.:
1. If the device (**`TS80P`**) was in **`soldering mode`*, the current tip temperature was **`410 °C`** and the tip was removed, it would have taken the device some time to reach the maximum temperature (here **`486 °C`**).
However, if it immediately jumps to this value, the **`soldering mode`** should be terminated.
2. It may also be worth checking if the temperature rises without user intervention to detect a removal of the tip.

Unfortunately I wasn't able to pull this of myself, so someone more capable might try that. 😊

discip, 2023-05-04:
@Ralim
As @alvinhochun pointed out, detecting a disconnected tip can fail due to the current method.

But what if the rate of temperature rise was used to determine this, since the temperature reading jumps to the maximum value almost immediately when the tip is removed?

e.g.:
1. If the device (**`TS80P`**) was in **`soldering mode`**, the current tip temperature was **`410 °C`** and the tip was removed, it would have taken the device some time to reach the maximum temperature (here **`486 °C`**).
However, if it immediately jumps to this value, the **`soldering mode`** should be terminated.
2. It may also be worth checking if the temperature rises without user intervention to detect a removal of the tip.

Unfortunately I wasn't able to pull this of myself, so someone more capable might try that. 😊

Ralim, 2023-05-16:
It's down to speed really 
It takes a while for the pull-up on some models to cause us to detect a removed tip. When we do it should exit soldering nice so you don't get hurt.

If you swap quickly then it won't see it.
A lot of soldering irons will great as soon as you change and I've often seen it as a good thing as I can just swap a tip and move on. But it does require one to be a little careful when handling the hot tips.

discip, 2023-05-17:
I'm not quite sure what you mean by that, as you added the **`enhancement`** label some time ago.

Is a reliable implementation of this feature request feasible?

If not, should this issue be closed?

jakobi, 2023-08-16:
It looks like I opened a duplicate of this in #1802 (as I searched for heat, not burned fingers :). 

Thx to @ia  for spotting it and pointing to the earlier issue issue. 

**Summary**:
- I have this issue with pinecil v1 and v2 under 1.21 (hakko is nice and slow  upto 24v, so not much of an issue if you've trained reflexes from soldering and avoid 6ohm tips :)).
- I have this issue with s60 (and here I really fear for the finger tips, 2ohm tip's a monster).

**And a new observation**:
If the old tip was sleeping, the new is not heating up. And only in this case pinecils and s60 detect tip removal.

Otherwise both only list the bogus temperature and continue heating with the switched tip. This happens with both heating up from lower temp as well after reaching the set temp. Regardless of fast switching or waiting a few seconds. 

**Workaround**: 
press 2nd button to make the iron cool done before switching tips.

**Implementation ideads (1ly distance):** 
- maybe add a boolean setting "tip change delay", and if a second has passed switch from assuming a flaky tip and misreadings to a tip must have changed.
- there's also the "heat-on-startup" setting which might also fit this case (assuming some kind of tip removal/change detection when actively heating/keeping the tip at set temperature)

jakobi, 2023-08-17:
Rethinking my previous post: shouldn't the title of this issue be changed to 

**cannot detect missing/broken tip when heating (incl. during tip switching)?**

Maybe also label it a **bug** now?