Ralim/IronOS #915: Feature suggestion: rectal thermometer mode

evils, 2021-04-01:
As we are in a global health crisis in which accurate core body temperature measurement is a useful diagnostic metric
and as [rectal thermometry](https://en.wikipedia.org/wiki/Rectal_thermometry) is the most accurate method of performing such a measurement,

I propose the addition of a rectal thermometry mode to IronOS.

### this feature must include:
* prominently displaying of the tip sensor temperature
* prevention of heating regardless of the available voltage and current
* prevention of entering this mode until the tip has cooled below 50°C
* a reminder to the user to not mix this mode with solder containing lead, and to clean off flux before use
* a disclaimer stating that this device is not intended for medical use

### this feature may include:
* additional calibration points at 0°C and 100°C for calibration with ice water and boiling water (can the iron boil the water for calibration?) (what about arbitrary calibration points?)
* additional displaying of the handle reference thermometer
  * regardless of a tip being present
* a visual indication that the measurement has reached thermal equilibrium
  * output on the pinecil breakout board for a piezo speaker for the same
* serial output via USB for continual measurement or remote indication of thermal equilibrium (as it can be hard to spot the display when using this mode alone)
* exclusion of this mode for irons that, for example, use T12 style tips without a flared base

### Acknowledgements
Using boiling water and core body temperature as calibration points has been previously suggested here:
https://github.com/Ralim/IronOS/issues/303
https://github.com/Ralim/IronOS/issues/303#issuecomment-388787802

DjordjeMandic, 2021-04-01:
Lmao

DjordjeMandic, 2021-04-01:
Imagine firmware bug where tip heats up without temperature/power limit while inside of anus lol

gmijo47, 2021-04-01:
Great idea for 1st April 😂😂

evils, 2021-04-01:
I agree that a lot of care will have to be taken to ensure such bugs do not occur.
I do not expect this to be a quick or easy feature to implement
and don't expect to see it until at least this time next year or later.

Additionally, some documentation warning about the use of a TS-K or [pine64 mjolnir](https://pine64.com/product/pinecil-mjolnir-hammer-head-perheater-tip) tip may be required.

DjordjeMandic, 2021-04-01:
No, I would like to see this implemented seriously :) Lmao

Ralim, 2021-04-04:
Honestly, my view on this is _whyyyy_.
The last think I would want people doing is putting a sharp object that is coated in metals you dont want contact with; in their rectum.

If you need this, just go buy a K thermocouple, or cooking temperature gauge at a pinch and use that, it will be more medically safe than this.

If you needed to do this, you mostly want to wind down temp setpoints and the power limiter (for safety).
Both the iron and the electronics are not designed to support measuring temperatures below around 5-10C above ambient, so as your hold the handle the unit will go outside of this and your temperature accuracy will rapidly decay.

DjordjeMandic, 2021-04-07:
[image]
😂😂😂😂😂

pfeerick, 2021-04-17:
Well played sir, well played! :) And just think of the health benefits! :laughing:

Techwizz-somboo, 2022-04-07:
I hate April fools, so many good ideas go to waste 😭

discip, 2022-04-21:
@evils
Since this was not intended to be taken seriously, please close this issue.

closed 2022-04-22 by Ralim