Ralim/IronOS #920: Notify translators to update translations?

alvinhochun, 2021-04-06:
_This is a question_

* **I have**
  - ✔️  Searched previous issues 
  - ✔️  This is in _this_ firmware, not vendor "offical" firmware
  - ✔️  This is not a bug in the vendor bootloader (aka DFU)
  - ✔️  I have checked this is not already covered in the docs in `/Documentation`

Since there had been quite a few strings added, how can we go about asking previous translators to update the translations?

Ralim, 2021-04-07:
You ask a very very good question that I dont yet have a good answer.

Currently its a bit of a self-serve situation which is not ideal; but I also dont particularly want to tag every language contributer :/

Do you personally have any ideas?

alvinhochun, 2021-04-07:
For the future maybe you can have a thread of some sort (maybe an issue) which translators can subscribe to, and announce to it when an update is needed (or just some time before a release)?

This doesn't solve the current update need however. For this time you may really need to ping them to get their attention.

unCleanCode, 2021-04-13:
I might suggest using online tools which provide this and even more functionalities, like poeditor.com

Ralim, 2021-04-25:
@unCleanCode I'm all for an online platform to do this, but most I have looked at require a bit of a mess of an API to integrate with the github actions workflow :(

playday3008, 2021-05-31:
crowdin, tranifex

closed 2022-05-08 by Ralim