Ralim/IronOS #934: Gloving red Pinecil

iLLiac4, 2021-04-16:
Today I wanted to solder something and the tip has become gloving red. I assume that the mosfet died.
Checking it I saw that it is Q7328. Where can I get a replacement for this mosfet? I have tride mosfet 4925 which is a good replacement for TS100 mosfet but with Pinecil it does not work.

TNX

Ralim, 2021-04-16:
Hia,

It's a common footprint so you should be able to source an equivalent part from digikey or mouser.

I would order from lcsc personally but depends where in the world you are (and you didn't say) 😅

iLLiac4, 2021-04-16:
Is there an alternative one that i can order from aliexpress?

alvinhochun, 2021-04-16:
I wonder if you really did kill the MOSFET. The CJQ7328 is rated for 8A continuous. If you were using any typical power supplies, a short in the output would probably shut down the power supply before being able to sink that much current.

> I have tride mosfet 4925 which is a good replacement for TS100 mosfet but with Pinecil it does not work.

With "4925" I found Vishay Si4925DDY and TSM4925D Dual P-Channel MOSFET, both of which looks to be pin-compatible, so I wonder how it didn't work.

iLLiac4, 2021-04-16:
You were right. 4925 mosfet works. I have replaced it with another one and obviously the first one was defekt. Now pinecil works again. TNX

closed 2021-04-20 by Ralim