Ralim/IronOS #936: Power Info Screen - DC/QC/PD

pfeerick, 2021-04-18:
_This is a Feature Request for TS80/TS80P/Pinecil_

To help with diagnosing some power supply and negotiation related issues, I thought it might be useful to have another debug or info screen that shows more than the PWR 0/1/2 info. 

i.e.
* in DC mode (PWR 0), say it's DC and just give the voltage. 
* in QC mode (PWR 1), say it's QC, what voltage was requested, and what voltage the PSU has given. 
* in PD mode (PWR 2), say it's PD, what voltage/current requested, and what voltage the supply gave, and reason for not getting voltage requested if it fails (i.e. when set/requested 12V, but 1.6A @ 12V is more than the 1.5A @ 12V the PSU can provide).

Thus at a glance you know which USB PSU mode the iron is in (DC is obvious, and only there for completeness) and can hopefully figure out why your nice 18W PSU giving you 12V via USB-PD even though you set IronOS to 12V... etc.

Ralim, 2022-02-15:
FYI, in you missed this.
The menu was updated ot show DC/QC/PD. Though we do not currently show the requested voltage for PD.

discip, 2022-04-21:
@pfeerick
> The menu was updated ot show DC/QC/PD. Though we do not currently show the requested voltage for PD.

Is this sufficient?
If so, please don't forget to close this issue. 😊

pfeerick, 2022-04-22:
Thanks for the reminder... I finally updated off of 2.15 (to 2.18) and I can see that there is better display of hte mode (PD, DC, etc) and voltage. The diagnostics part is missing, but I thought that would be a stretch, as this is a small screen device and I don't know how you guys are going for firmware/feature space. So I will consider this closed, and leave it to someone else to push for that if they wish.

closed 2022-04-22 by pfeerick

Ralim, 2022-04-22:
If you have a Pinecil and you boot it up holding the front button from a USB-C source; it will bring up a menu similar to the debug menu that shows you what the power adapter is advertising for specs.

pfeerick, 2022-04-22:
*I take that back... sorry!!* Yes, that is exactly the sort of thing I was suggesting... thanks! 😄