Ralim/IronOS #938: TS100 Only DFU Mode

JoaoAragaoVC, 2021-04-19:
Hello everyone, 

Today it came my ts100 and the first thing i've done was to flash this Ralim Firmware, well no luck for me. I'm on Windows, and the TS100 is on DFU 3.45.

The only thing I can do with this TS100 is enter the DFU, nothing more. When I plug in DC power, nothing appears on the display. 
When I enter DFU Mode, the windows detect new device, pop up the *empty* folder and then I try to drag the hex file and the file transfer is NOT concluded. The device resets almost immediately. After the reset it detects device but don't show any device folder. 
Then I take the usb cable out and go to DFU again, folder empty again, and so on and so forth. 

I don't know my FW version, all I know is DFU 3.45 because it's all I can see. 

Does anyone of you have experienced this? Should I ask for a replacement? 

Thanks in advance!

krisiasty, 2021-04-19:
Which exactly hex file are you trying to load?
Have you tried to load original firmware? You can find it here: https://www.minidso.com/forum.php?mod=viewthread&tid=892
If that doesn't work, try another USB port (some people reported issues when using USB3.0 port). different cable or/and different computer.

JoaoAragaoVC, 2021-04-19:
Hello and thanks, I've used the last one of Ralim and the original one. The weird part are that the folder is empty when I load dfu mode, and when i begin the file transfer the device automatically resets not concluding the file transfer.

mullcom, 2021-04-19:
I did flash My ts100 today. Att firar i did not get any respond in my
windows 10 (no device)

I solved this by changed USB cable. Some is only power USB and don't have
full support for data connection.

The other thing I did was download a driver. I did some google find a
program that did had this driver.

I have latest framware now. 👍

send from mobile

JoaoAragaoVC, 2021-04-19:
I've sucessfully transfered now changing the pc and usb cable. Facepalm. Thank you, can I close this issue?

closed 2021-04-20 by Ralim

Kabloempia, 2022-09-23:
> Ralim

Hello, What driver did you use to update your ts100? I cannot find anything that seems to work.

Kabloempia, 2022-09-23:
> Hello everyone,
>
> Today it came my ts100 and the first thing i've done was to flash this Ralim Firmware, well no luck for me. I'm on Windows, and the TS100 is on DFU 3.45.
>
> The only thing I can do [...]

Hello Joao,

I have exectly the same issue as you right now. 

1. tried to update the ts100 by pasting the ts100 hex file in the device using usb. 
2. the  device disconnected and connected again 
3. device did not updat and now the device does not turn on anymore when I put it on the dc connector. just a black screen 
4. Device now only powers on when it is put in DFU mode. 

At the end of your post you mention that it is solved but how did you solve this problem? 
I only got to toe point where the ts100 was updated with the correct file and the file changed to ready after disconnecting and connecting again. 
But the ts100 still wont turn on when I put it on the dc. No response to anything. 

How did you get your ts100 to turn on again? 

Thanks in advance if you be able to help me out!

discip, 2022-09-23:
@Kabloempia
> changing the pc and usb cable

That apparently solved the problem for @JoaoAragaoVC.