Ralim/IronOS #942: Detect constant tip temperature when in heating mode and stop after a short while

gdonval, 2021-04-28:
_This is a Feature Request_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

If, for whatever reason (most likely hardware failure, like fried op amp), the reported tip temperature is spurious and stays low, the iron will heat like there is no tomorrow.

* **What is the expected behavior?**

The iron should not dangerously overheat ever. If something fishy is going on (like "staying" at 23°C for 10s while heating at 65W), the iron should go into safe mode (i.e. shut down the tip) and report the error to the user.

Different tips have different thermal behaviour so just checking that steady state has been reached in X seconds is probably not a good idea. In terms of PID parameters, P should decrease at a reasonable pace at first: if it stays big and constant (i.e. D is very small), there is a problem. When P is small enough, we are at temperature.

* **What is the motivation / use case for changing the behavior?**

The current behaviour is dangerous and defeats most of the other safety measures in place (sleep mode, etc.) because from the iron perspective, the tip is and stays cold while it's almost melting in real life.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.14
  - PCB Version: (1/2)
  - Power Supply: PinePower Desktop (USB-C, 65W, 20V, ~3A)

ssl-3, 2021-06-05:
I came here to make this suggestion, having found a related complaint in the Pine forums:  https://forum.pine64.org/showthread.php?tid=13585

I would like to add that this problem has been solved in the Marlin 3D printer firmware with their "thermal runaway protection" system.  Marlin is capable of detecting slower-than-sane heating, as well, which also catches the failure modes where the sensor is working but is poorly-coupled to the heater itself.

Perhaps some ideas or code could be borrowed from there.  

It seems like a preventable failure mode and there's certainly plenty of CPU grunt available to make it work.

Ralim, 2021-06-05:
I've put a rough punt at a catch for this into #993.
If you have a unit that can replicate this, can you please test?

> I came here to make this suggestion, having found a related complaint in the Pine forums: https://forum.pine64.org/showthread.php?tid=13585
>
> I would like to add that this problem has been solved in [...]

You are most welcome to open a PR bringing across the logic if you would like.  Last I looked at Marlin it was a moderately dodgy Arduino sketch that had this annoying feature where it would kill my bed heater as it didnt heat fast enough, that I spent an hour patching out 🤣 . But naah its good firmware but also not the nicest to extract logic from.

> 
> It seems like a preventable failure mode and there's certainly plenty of CPU grunt available to make it work.

Absolutely, but I'm not paid anything to work on this firmware remember, so the amount features is proportional to my free time and willpower :)

PR #993 (2021-06-24): Thermal Runaway alert
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [ ] The changes have been tested locally
- [ ] There are no breaking changes

* **What kind of change does this PR introduce?**
Trying to fix #942 

* **What is the current behavior?**
Unit does not detect thermal runaway and turns tip into glowing nightlight

* **What is the new behavior (if this is a feature change)?**
This _should_ throw up an alert and kick the user back to the main menu

* **Other informatio [... cut]
closed 2021-06-24 by Ralim