Ralim/IronOS #948: Translation updates subscription topic

Ralim, 2021-05-01:
Hello,

To save doing callouts for translation updates via releases; going to try having everyone who would like to be notified when there is going to be a release so that they can do a translations update pass to please comment in this issue.

Before doing a release I will tag everyone here to send an alert that there is an upcoming release and we would love all of your lovely people to update translations.

If you would like to stop being notified, you can later on delete your comment.

Please no other off topic chat in this channel. I will delete off topic comments :)

So if you would like to be notified please comment with the language(s) you speak for all of us to be aware :)

PS: Sadly I'm Australian so I really only speak Aussie English; thus why I dont even attempt to touch other languages 🤟🏼

Vinigas, 2021-05-01:
Lithuanian

HanaO00, 2021-05-01:
Hello,
It's a good idea. I help with the French translation.
TS100 - Pinecil.

federicodilo, 2021-05-02:
I'll take care of the Italian translation

discip, 2021-05-04:
Good evening sir,
as long as I am able to, I could do the ***German*** and ***English*** (if needed at all) translations.

IkerGarcia, 2021-05-06:
Hi,

I'll do the Spanish one if you want. In the past I've translated the Pi Pinout.

hippi-viking, 2021-05-06:
Hello :)
I'll try to help out with the _Hungarian_ translation.

Dag0d, 2021-05-09:
Hello,

Happy to do the **_German_** translation

hrebeniukB, 2021-05-20:
Can make Ukrainian translation, if u need

piksel, 2021-05-25:
I'll try to update the **_Swedish_** translation then.

KITT3000, 2021-05-27:
I can update the Polish translation.

cipyyy, 2021-08-11:
Hello,
I have attached the file translated into Romanian.
I don't know how to compile to get firmware.
Please help me, I promise to update the file for the Romanian language whenever needed.
[translation_RO.zip](https://github.com/Ralim/IronOS/files/6970373/translation_RO.zip)

playday3008, 2021-09-26:
Ukrainian (mostly), Polish, Russian

onxblog, 2021-11-19:
Hello,

Happy to do the Russian translation :)

singhWeb, 2021-11-29:
I'll take care of the Hindi and Punjabi

0Karakurt0, 2025-07-23:
I'll add to Ukranian translation, though I don't expect there would be big changes to the strings)