Ralim/IronOS #949: Round two USB-PD

Ralim, 2021-05-01:
As the latest PD update improved support for Apple chargers but made the situation worse for other adapters; going to track here progress and attempts to resolve this.

Plan here is roughly:

As I work on the MHP30 I'm doing a bit of a general cleanup of the PD stack while I'm at it.
Will post here with versions of the firmware that I would like to have tested to check if things get better or worse.
Before asking here I will have tested on the four adapters I have to hand.

Ralim, 2021-05-01:
First test idea is this version: https://github.com/Ralim/IronOS/actions/runs/801274341

@xxv would be curious if this helps you.

Vinigas, 2021-05-01:
> First test idea is this version: https://github.com/Ralim/IronOS/actions/runs/801274341
> 
> @xxv would be curious if this helps you.

Feedback: This build fixed PD issues with current v2.15RC1 and even fixed issue with Lenovo loptop PD charger from v2.14.1 bug. Meanwhile MHP30 failed to flash ( `MHP30_EN.NOT` filename).

Ralim, 2021-05-01:
@Vinigas 
Fantastic to hear that helps. I think the main issue that was being run into was the fifo filling up with messages as the IRQ turnaround was not quick enough. (Especially when the TS80P uses bit-banged I2C).

Only looking at PD from that branch at this point in time (for this issue). The MHP30 firmware is not setup to be flashable via the Miniware DFU (yet).

elschopi, 2021-05-02:
Feedback from my side: (TS80P, Anker 30W wall plug and 30W PD powerbank)
same problems as with the pre-release2.15 RC1: TS80P negotiates only 9V. With release 2.14.1 it manages to negotiate 12V 30W at least on the powerbank. 
Either my iron or my powerbank / charger are haunted it seems :D

Ralim, 2021-05-03:
> Feedback from my side: (TS80P, Anker 30W wall plug and 30W PD powerbank)
> same problems as with the pre-release2.15 RC1: TS80P negotiates only 9V. With release 2.14.1 it manages to negotiate 12V 30W at least on the powerbank. 
> Either my iron or my powerbank / charger are haunted it seems :D

This is because your power bank is 30W and not the 32W+ required 🤣 but I do intend to be a tad more tolerant as to what is negotiable on PD.

Ralim, 2021-05-04:
@elschopi 
Could you give https://github.com/Ralim/IronOS/actions/runs/809565338 a try?

xxv, 2021-05-04:
> First test idea is this version: https://github.com/Ralim/IronOS/actions/runs/801274341
> 
> @xxv would be curious if this helps you.

Woo! This made the Lenovo PSU work properly (negotiated 9V, 1st boot). Stock charger too (12v, 1st boot). Both Anker PSUs. So far it seems like this update has fixed all the issues I had been experiencing. Thanks so much.

I did notice that on one of the Anker PSUs, it would occasionally (1 of 10 times) only give 5V (1st boot), but unplugging and replugging would fix it. It may have been when I unplugged / replugged quickly. I can provide more details and try to make it reproducable if that'd be useful to you.

Ralim, 2021-05-04:
I'm very glad to hear it has helped 🎉 

If its a quick re-plug that may be the charger going into safe mode. But if you do come across a way to trigger it reliably I am all ears and keen to hear about it :)

elschopi, 2021-05-05:
> @elschopi
> Could you give https://github.com/Ralim/IronOS/actions/runs/809565338 a try?

sorry for the late reply. I'll give it a try later today when I'm back from work. Thank you nonetheless =)

elschopi, 2021-05-06:
> @elschopi
> Could you give https://github.com/Ralim/IronOS/actions/runs/809565338 a try?

So, finally got to test it yesterday evening 😅 
This version behaves just like the release 2.14: the 30W powerbank negotiates 30W, the Anker wall plug just gives 9V at 18W. 
I've also tried to quickly disconnect and reconnect the Anker, but that didn't do anything for me. 😄

limited660, 2021-05-10:
Hello, I have a TS80 on 2.07 that can get 12v but a Pinecil on 2.15.40087E6 cannot get above 5v on the same powerbank, cable, and port. Powerbank is an Anker PowerCore Speed 20000 Model A1278. Any ideas?

Ralim, 2021-05-10:
@limited660 that power bank only supports QC3 so performance on Pinecil will be fairly poor.
I would suggest following up in the #823  thread as its a QC issue and nothing to do with USB-PD

closed 2021-05-11 by Ralim