Ralim/IronOS #965: MHP30 doesn't heat above ~70-90°C

Avamander, 2021-05-10:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a Bug_

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [X] This is not a bug in the vendor bootloader (aka DFU)
  - [X] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
* Doesn't heat above ~70-90°C

* **What is the expected behavior?**
* Heats up to specified temperature

***Steps to reproduce the bug:***

- Turn on...
- Press to heat up

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.15.40087E6
  - PCB Version: ?
  - Power Supply (Voltage and Current Rating): Lenovo SA10R 65W (20V@3.25A, 15V@3A, 9V@2A, 5V@2A)

Vinigas, 2021-05-10:
How many Watts does it use at peak (enable detailed heating screen via settings to see) ?

What voltage have MHP30 negotiated with this charger ?

What Power Limit value is set in settings ?

Avamander, 2021-05-10:
> How many Watts does it use at peak (enable detailed heating screen via settings to see) ?

63.9W

> What voltage have MHP30 negotiated with this charger ?

20.2V

> What Power Limit value is set in settings ?

65W

> Have you tried factory reset after flashing ?

I have now, issue persists.

Ralim, 2021-05-10:
> Also, fun fact, flashing only happens if you use the name of the HEX the manufacturer uses.

This has never been an issue for me, you may be running into the double-tap flashing quirk here.

I know its tedious but can you dump out the contents of the debug menu values for reference?
I can't see any easy reason why this would occur off the top of my head.

Avamander, 2021-05-11:
> This has never been an issue for me, you may be running into the double-tap flashing quirk here.

I tried multiple times on Windows 10 to copy-paste the firmware file to the DFU storage, it ended up with ERR most of the times. The time I renamed the file, it flashed instantly. Could've just been a massive coincidence.

Avamander, 2021-05-11:
>  I know its tedious but can you dump out the contents of the debug menu values for reference? I can't see any easy reason why this would occur off the top of my head.

Debug menu?

closed 2022-05-08 by Ralim

KayThy, 2022-08-22:
I am facing the same problem but my hot plate only get up to 27 degrees and doesn’t increase afterwards. What must I do? I have never used hot plate before so I’m not sure where to go from gere