Ralim/IronOS #969: more realistic animation of the wrench

discip, 2021-05-14:
> While you are at it, would you mind to change the animation style oft the _**wrench**_, used for the **`Advanced options`** menu, as well?
> I thought about the movement of the real thing _(rotating around one end)_, not sliding in. 😁

IMO better handle that in another issue.

_Originally posted by @alvinhochun in https://github.com/Ralim/IronOS/issues/892#issuecomment-831778208_

federicodilo, 2021-05-14:
It is not easy to draw a better wrench in a 16 × 16 square, or at least one that is recognizable.
Furthermore, as it is drawn diagonally, it must be made smaller and therefore less "legible".

discip, 2021-05-14:
The idea is not to depict the whole wrench at any point of the animation, rather than seeing it rotate, even if parts of it get cut away in the process.
I tried to create a sample of how it could look like. 😅
[image]
<details>
  <summary>
    <b>individual frames</b>
  </summary>

1. [image]
2. [image]
3. same as the 1st
4. [image]

</details>

regards

HanaO00, 2021-05-14:
Hello, why not put a spinning gear? Also, it will be a little more familiar for some people compared to the settings of the phones.
Something like this
[image]

discip, 2021-05-14:
@HanaO00
1. The wrench was originally there, it just should look the right way. 😊
2. As @federicodilo stated, there is only a
> 16 × 16 square

and only ***3 frames*** where this delicate animation (2 meshing gears) has to fit into.
The .gif provided above, is actually fulfilling these requirements, but I doubt these gears will. 🤷‍♂️

HanaO00, 2021-05-14:
Ah, I didn't know that there were only 3 frames, I'm only offering a few ideas, that may give you some too, I found this, It is also possible to play with a zoom / dezoom maybe (smaller to bigger)
[image]

federicodilo, 2021-05-15:
Here I have two new proposals:
[image] [image]

The current animation of the two thunders always give me the impression to be "glitchy".
Whatever, let's try to find a compromise.

HanaO00, 2021-05-15:
The lightning could be confused with the one present in the power settings, maybe?
Difficult to see how it looks on the soldering iron screen, better than here I imagine, the variant is very interesting, an additional idea, there is this type of key too, which can have an opening movement and closing
[image]

federicodilo, 2021-05-15:
Yes, sorry, the lightning it's intended to possibly replace the animation in power settings.

HanaO00, 2021-05-15:
Oh yes sorry I read it wrong, I like the new proposal of the lightning

HanaO00, 2021-05-15:
Oh yes sorry I read it wrong, I like the new proposal of the lightning
I confirm for the second variant
For the last one wrench, maybe reverse the animation (opening, there is closing)

federicodilo, 2021-05-15:
Each gif is designed so that the last frame retains a certain harmony so as to be "readable" if the animations are deactivated.
As for the wrench I think it is better to have it less open in the last frame.

HanaO00, 2021-05-15:
Interesting, thank you for all this info

Vinigas, 2021-05-15:
[image]
What about this ?

HanaO00, 2021-05-15:
I prefer the one made by federicodilo which does not go beyond the frame personally

Vinigas, 2021-05-16:
[image]
v2. Opinions ?

HanaO00, 2021-05-16:
As federicodilo said there are 3 frames in the gif, I like the very last one

discip, 2021-05-18:
Good morning everyone,
as @federicodilo mentioned above:

> Each gif is designed so that the last frame retains a certain harmony so as to be "readable" if the animations are deactivated.

and even before that, these annotations were addressing the same issue:

> 1. [image] _has to be the **first**_
> 2. [image]
> 3. [image] _hast to be the **last**_

the last frame has to be the most *representable*.

And as the original design is the one, which should be kept, since the initial idea was not to replace the icon, but rather to make it look natural, remember? 😁
My suggestion is to go with the original wrench, but slightly shortened to fit the whole tool in every frame, make it go clockwise and place the most "readable" frame at the end, for cases where animation or looping is turned off:

[image]

<details>
  <summary>
    <b>don't click</b>
  </summary>

just in case
1. [image]
2. [image]
3. [image]

</details>

kind regards

discip, 2021-05-22:
@Ralim
@alvinhochun
If this is not to hard and you agree with this change, it would be nice if someone could implement this.

thanks
kind regards

closed 2021-09-13 by discip