Ralim/IronOS #975: USB-PD Problem

HanaO00, 2021-05-21:
Hello, I am using a PinePower power supply, to power the Pinecil, however, on my two Pinecil I have a shutdown when the soldering tin hits the tip, and I don't have this problem using the power jack, is this problem is firmware, or it comes from the power supply? I put two videos to illustrate
Thank you

https://user-images.githubusercontent.com/17810985/119063233-e3272500-b9d8-11eb-8cc3-9934bb279c49.mp4

https://user-images.githubusercontent.com/17810985/119063217-dc001700-b9d8-11eb-81a8-32b499cd958f.mp4

Ralim, 2021-05-21:
Hmm this is interesting.
I agree with your edit that grounding would be the best solution.
I would hesitate to guess this is to do with static or grounding. As the pine power units are double insulated and not grounded, the output is floating. If you were to be grounded and provide a ground path, I could see this tripping a safety somewhere.

Not honestly sure what is best here other than adding a ground connection.

alvinhochun, 2021-05-21:
Can you try flipping the C7 (figure 8) connector of the power cord connected to the PinePower? Swapping the mains Live and Neutral input might change something.

HanaO00, 2021-05-21:
@Ralim Thanks for your reply, so do you think that if I do one feed referral and get another one, I would have the same problem ?

@alvinhochun Thanks, I have reversed the C7 connector on the power side, so far I haven't had the problem yet, but I don't understand, isn't it the same to reverse the connector on the wall side too?

And last question, do you think it damaged my soldering irons?

alvinhochun, 2021-05-22:
> I have reversed the C7 connector on the power side, so far I haven't had the problem yet, but I don't understand, isn't it the same to reverse the connector on the wall side too?

It's the same to flip the wall plug, but I didn't want to assume that you are using a symmetrical wall plug.

HanaO00, 2021-05-22:
Ah yes sorry I misunderstood, strange for the moment I have not yet had the problem, and I tried to see the voltage in usb it seems OK
[image]

Ralim, 2021-05-25:
If flipping the plug has solved the issue I would be suspect of issues with leakage between the power supplies input and output.
A grounded supply would solve this issue, but otherwise you may just need to know what orientation works best.

Not overly sure what else you could do here, as if your tripping the supply the iron can't do too much about it.

HanaO00, 2021-05-25:
Thank you for taking the time to answer me, I don't know if it's the placebo effect, but I haven't actually had a problem since, if you would have been in my place would you have changed the power supply?
If it does not risk anything for the soldering irons it is good, for the grounding knowing that the power supply does not have any I have to use the crocodile clip
You can close the topic whenever you want to avoid polluting the issues, thank you very much for your help

Ralim, 2021-06-02:
I'm a big believer in grounding supplies, but its a bit 🤷🏼 if you would need to in this case.
I would probably keep using it but try not to use it on sensitive electronics :/

closed 2021-06-02 by Ralim

timmoeh, 2022-03-04:
Hi, I have the same problem with my two pinecil when I‘m using them in combination with my pine power desktop. I still tried to flipp the C7 connector like alvinhochun propose. But this doesn‘t works for me. Are there any other ideas to fix the problem? Thanks for the help!

Ralim, 2022-03-06:
@timmoeh This ticket is closed, please open a discussion question or ticket or use the Pinecil community chat for this.