Ralim/IronOS #980: TS100 Display Issues

wwilkinson192, 2021-05-24:
I do not think this is an issue with your firmware.

But I am hoping you can help.

I have a new TS100 It has always had issues with the display messing up.
DFU 3.45 Miniwave  Firmware ver 2.18

I thought that it would go away with the firmware upgrade but the display is just blank.
Sometimes I can get it to heat but nothing on the display with your firmware.
I have seen some issues close to this but no fix.

What information can I send you to get this working?

Your Firmware is the only thing this is missing. 

Let me know.
Please!
Thank you in advance.
Wayne

Ralim, 2021-05-25:
Hia,
If you are having issues with the display messing up, return the unit as defective and have a replacement supplied.

That is your best bet, as this firmware doesnt really support faulty hardware. The Miniware firmware is slightly more tolerant of issues due to its design of using software implementations of hardware protocols. But as these come at the cost of speed this firmware avoids them were possible.

Your only other alternative would be to desolder the screen and attach a new one, which is not trivial to do.

ricktendo, 2021-10-11:
The OLED screens burn out quite easily, like Ralim said send it back, or you (or somebody) can replace the screen, you will need to source a white 0.69 inch screen (ebay, aliexpress, bangood, etc.)

discip, 2022-03-31:
@wwilkinson192
Since you stated your self, that this is not firmware related and I assume you got the anticipated help as well, please close this issue. 😊

thanks

closed 2022-05-08 by Ralim