Ralim/IronOS #983: Display sleep

vladkozlov69, 2021-05-27:
**Is your feature request related to a problem? Please describe.**

A common issue with TS100 and TS80 is OLED degradation. 

**Describe the solution you'd like**

It would be great to have display sleep mode, i.e. turning display off after 1 minute (could be configurable) and turning it on when button is clicked.

Ralim, 2021-05-27:
This already exists then the device is not in use :)
If you leave it to sit it will go to sleep

discip, 2021-05-27:
@vladkozlov69
Very nice some one else, who is interested in sparing his OLED. 😁👍
## But there is a more elaborate way of achieving this proposed by @ceut2k4 here https://github.com/Ralim/IronOS/discussions/836.
Maybe you could upvote this as well to make it get the deserved approval.

regards

vladkozlov69, 2021-05-28:
> @vladkozlov69
> Very nice!
> Someone else, who is interested in sparing his OLED. 😁👍
> 
> ## But there is a more elaborate way of achieving this, proposed by @ceut2k4 here #836.
> Maybe you could upvote this as well to make it get the deserved approval.
> 
> regards

That's a good idea as well, I think we can combine both and make display contrast more "smart" based on device usage.

Ralim, 2021-05-28:
You are aware that on these cheap oleds using them degrades them around the same or less than leaving them off in a drawer?

I've had more die unused in storage than due in use. :)

discip, 2021-05-28:
@Ralim
Good evening sir,

> You are aware that on these cheap oleds using them degrades them around the same or less than leaving them off in a drawer?

No this one is new to me! 😲
But if the request of @ceut2k4 is not to hard to implement, it either way would be a great option.

IMHO:
IronOS stands out at its current state already, but would approve its greatness even more, if this option would be added too.

Anyway:
Thank you for your effort so far. 😃👍

kind regards

discip, 2021-07-16:
Good news,
@alvinhochun created a PR wich could be the solution to this, if someone would make it persistent! 😁

regards

discip, 2022-03-31:
@vladkozlov69
Do you still need this to be realized, since a *'brightness-setting-option'* was implemented?
If the 'brightness setting' is sufficient, please close this issue. 😊

thanks

closed 2022-04-01 by vladkozlov69