Ralim/IronOS #984: sequre sq-001

bm16ton, 2021-05-28:
Hello, this project is amazing! I have a Sequre SQ-001 which appears to be a TS1000 clone. I found this project in a review for my iron here https://oscarliang.com/sequre-q-001/ He claims to be using this firmware on the SQ-001  without problems. Just curious if you have heard of the SQ-001 and if plans to officially support/list it. Also if i take the chance and flash iron IronOS what possible issues should i be aware of, and/or any kinda tests I should do? Thanks you! Again awesome and super impressive work!

Ralim, 2021-05-29:
Hi,

Everything _should_ work totally fine as they are a clone. Just don't list them officially as the manufacturer tends to have poor support if they fail for one reason or another and trying not to encourage the clone market :)

That said, it's a hardware copy so everything should work and act like the Ts100.

Have fun, and may it work well for you

closed 2021-06-05 by Ralim

bm16ton, 2021-06-09:
Well for the most part it works! But it does lock up a fair amount. I have been running it with the extra info on home screen but will try it wo that setting. Have tried multiple psu's so dont  think they are the issue. Will test more as I can.