Ralim/IronOS #985: Update docs for perm vfat msdos fix?

bm16ton, 2021-05-29:
rather then remounting msdos manually everytime or run a script , id prefer to have it automatically do this so drag and drop just works, also set permissions for write access. 
edit /etc/fstab

/dev/disk/by-id/usb-Virtual_DFU_Disk-0:0 /mnt/usb-Virtual_DFU_Disk-0:0 msdos nosuid,nodev,uid=1000,gid=1000,nofail,noauto,x-gvfs-show 0 0

It now automounts correctly on plugin. I understand why holding a was needed when we had two options config and dfu, we needed a way to select between the two, but is it needed anymore? Any other reasons for plugging into usb port then dfu? If not can we remove the A button requirement?

Ralim, 2021-05-30:
Happy to add this to docs or you are welcome to open a pr.

I cant change the button holding requirement as that is part of the vendor firmware and that cant be changed without rewriting their bootloader.

River-Mochi, 2023-03-23:
has this been completed, adding this advice  to the Documents,  so this ticket could be closed from 2 years ago?

closed 2023-03-29 by discip