Ralim/IronOS #992: Auto reset after achiving set temperature

DarkEaglous, 2021-06-05:
Hello,
I have just bought TS100 and i have updated firmware to v 2.15.40087E6  09-05-21. 
Solder heats up nicely, but when it reaches desired temperature, it resets itself and temperature starts dropping... When i set solder to start heating up right after plugging in, it reaches temperature, resets, heats, resets, heats and so on... I've tried more language pack if that is not only problem in my language but unfortunetly not.

Do any of you have any idea what could be wrong? thank you very much

Ralim, 2021-06-05:
Hello,

Can you provide the missing information you took out of the issue template:

**Details on your device:**
 - Device: [e.g. TS80/Pinecil etc]
 - Release: [eg 2.15.40087E6]
 - Power adapter being used:
 
 Especially we require details on how you are powering the device as that's usually the number 1 suspect for these sorts of issues.

DarkEaglous, 2021-06-06:
Hello,
Device: TS100
Release: 2.15.40087E6 09-05-21
Power adapter being used: Multiple original Notebook power supplies - ASUS, Fujitsu Siemens, Acer (19V 4A). 

I have tried more options today wit power sources since you said it is problem no.1 and i have found out that TS100 works properly only if i connect it on xys3580 DC-DC power supply (powered by mentioned notebook power supplies). I have no idea why TS100 doesnt like output of my Notebook power supplies, but thanks to your advice i kept digging in possibilities and in the end i have found working solution.
Thank you very much sir.

Ralim, 2021-06-08:
No problem at all 🤗 
Glad it helped :)

Its a confusing one that shows up occasionally, generally due to power supplies not being used to a load that can go from 0% to 100% fast and back again. Some power supplies just don't like the oscillation.

closed 2021-06-08 by Ralim