Ralim/IronOS #995: Pinecil shows 20-30 C while the tip melts solder

skorokithakis, 2021-06-08:
I'm using the latest Pinecil firmware for a Pinecil I got today. I'm powering it from my Lenovo laptop's USB-C adapter. When I start heating, the iron keeps showing 20-30 C while the tip is so hot it can melt solder (I didn't keep it powered for longer than a few seconds, when I started smelling the "hot tip" smell).

When it starts cooling down, the numbers vary from 100 C to 450 C from one second to the next. It seems like there's something very badly wrong with the sensor.

I ran a temperature calibration and got 2234 as the value. What could be wrong?

Ralim, 2021-06-08:
Most likely issue is the op-amp circuitry inside the Pinecil is non functional.
There is no firmware fix for this issue. This will require hardware repair :(

However, as you are here could you do me a massive favour and test if this stops you with an alert for thermal runaway?
https://github.com/Ralim/IronOS/pull/993

skorokithakis, 2021-06-08:
I sure could and will. Do you want me to just run it and see?

Also, this is a brand new iron, have you heard of this happening often?

Ralim, 2021-06-08:
Just run it and see, when you start heating after a bit it should turn the heater back off and show the warning.

There have been a couple but its certainly in the < 1% of cases.
You will need to file a ticket to Pine64 around it and they should replace the unit

skorokithakis, 2021-06-08:
Testing it, it doesn't seem to stop. I don't know how hot it gets, due to the hardware issue, but it definitely melts solder easily. I'm afraid to leave it for much longer.

skorokithakis, 2021-06-08:
Do you know which opamp they use? Maybe I can solder a new one myself.

Neodymium5312, 2021-06-12:
> 
> 
> Do you know which opamp they use? Maybe I can solder a new one myself.

I was just browsing the issues threads and figured since I know where to find this i'd chime in,
I'd see if they will replace your unit through warranty/support but if you want to do it yourself the wiki lists most of the user replaceable components.

according to the pinecil wiki its this component:

SGMicro SGM8557-1 Low Noise OP Amp
Per: https://wiki.pine64.org/wiki/Pinecil#Datasheets_for_components_and_peripherals

skorokithakis, 2021-06-12:
Thank you, they're already sending me a replacement but I searched for that component and couldn't find one for sale anywhere. Only similar models were available, which I don't know if they're compatible...

Neodymium5312, 2021-06-12:
> 
> 
> Thank you, they're already sending me a replacement but I searched for that component and couldn't find one for sale anywhere. Only similar models were available, which I don't know if they're compatible...

Sounds good 👍  
Should probably close out this issue now that its sorted
FFR, you could have also used the datasheet to find a working replacement with the same specs but 🤷 guess a working iron is all that matters.

skorokithakis, 2021-06-12:
Hmm I'm not that good at electronics to tell whether it'll work, but I might give it a shot, thanks!

closed 2021-06-12 by skorokithakis

Ralim, 2021-06-14:
@skorokithakis 
You can use most generic op-amps that are 20V tolerant and same pinout and it should work totally fine.
If you replace it and you still have issues feel free to open an issue to discuss.

skorokithakis, 2021-06-14:
@Ralim fantastic, I'll try to get some of those! Would you happen to know a good/reliable part with the same package offhand? If not, I can look around.

Ralim, 2021-06-14:
@skorokithakis 
Sadly dont know any off the top of my head, though basically any with the same pinout that can run at 3.3V or more should work.
We are not really pushing any of the specs as far as I know, though choose low noise where possible.
Depending on your source of choice,I would just pick up one or two cheapish models and try them out.

skorokithakis, 2021-06-14:
Great, thanks!