Ralim/IronOS #996: Ts80p temperature problem.

12321312, 2021-06-09:
Hello.  I am having some problems with the temperature reading on the displays.  If I use a TS-B02 tip and set the temperature to 330 degrees, then on the displays it jumps sharply from 330 to 340, and the readings change abruptly, not static.
 On the TS-K4 sting, if I set the temperature to 330, then the readings change from 320 to 370 there back, not static.

Ralim, 2021-06-09:
Hello, please add back in the missing information block that you have not answered about your model and power supply :)

12321312, 2021-06-09:
TS80P ironos 
30W original

Ralim, 2021-06-09:
What model number, what voltage, what firmware version.
firmware version should be 2.15.XXXXXYYYY when you read it off the device.

Are your tips the official ones from Miniware (have logo laser engraved on them) ?

12321312, 2021-06-09:
[image]
[image]
[image]
[image]

12321312, 2021-06-09:
Was stitched to the official firmware, there is no such problem.  The problem is on this firmware.

12321312, 2021-06-09:
https://youtube.com/shorts/-9s33oEp7Lw?feature=share
 More clearly

Ralim, 2021-06-09:
Hi,
Thank you for providing the firmware version for confirmation. ❤️ 

Looking at the video it looks like you are seeing the normal movement of the last digit that is normal as that is the firmware showing "live" the exact reading. (Rather than hiding this like official firmware).

There is already a ticket about adding a filter to hide this but this has not yet been implemented.

Also you can try using it with the power pulses turned off as these can make this situation worse as the firmware inserts small pulses of power to keep power banks awake, but this comes at the detriment of the temperature regulation.

12321312, 2021-06-09:
Thanks for explaining.  I thought it was a marriage and was very sad.  How do I turn off impulses?  Can I have a mini-guide?

Ralim, 2021-06-09:
I will also add the movement of the numbers are of no detriment to performance. (just ugly)

I believe if you go into the menu -> Advanced settings -> Power pulse you should be able to set it to be turned off. (Menu names depending on translation)

12321312, 2021-06-09:
Thank you very much.  I will be looking forward to an update where this feature will be.  :)

closed 2021-06-10 by LousyFox