Ralim/IronOS #1000: 'Not Enough Disk Space' When Mounting TS80P

mmehrle, 2021-06-22:
I tried this on OSX Big Sur as well as Windows 10. When mounting the iron per these instructions:

https://github.com/Ralim/IronOS/blob/master/Documentation/Flashing.md

... I get the following error message:

https://i.postimg.cc/bNdfYtVD/image.png

'Not enough disk space'. Which is strange... anyone ever run into this with the TS80P?

Ralim, 2021-06-22:
Almost 100% you are not entering into DFU mode.
The screen should say `DFU` on it, if it doesn't you are not in DFU mode.
Also, it looks like you _might_ be using a VM. This **will** cause issues. 
Just use your host machine.

mmehrle, 2021-06-22:
I just tried it again and suddenly it showed up as a proper drive. Then it worked. So I guess this may be the shortest lasting issue in Github history ;-)

closed 2021-06-22 by mmehrle