Ralim/IronOS #1001: Pinecil Hall Sensor

mikerocklewitz, 2021-06-22:
Hey all,

I ordered some hall sensors off of d-key to add the magnetic shutoff to my pinecils.  I followed the schematic from the pine64 wiki and ordered the exact Si7210-B-00-IV(R) as listed.  The installation was pretty straightforward - 5 points to solder.

But....

How do i make it work? It makes no difference what sensitivity I set it to, it does not detect magnetics.

[image]

Ralim, 2021-06-22:
Hi,

If the menu setting is showing the sensor is being detected.

Make sure your magnets are _strong_.
I use a large neodymium one mounted near the iron when in the stand.

mikerocklewitz, 2021-06-22:
The only setting for the hall sensor that I see is sensitivity under sleep settings.

I assume that's the only setting.

mikerocklewitz, 2021-06-22:
Hah.

it works fine.  Apparently knowing where the magnets are located in the soldering stand is pretty important.

closed 2021-06-22 by mikerocklewitz

discip, 2022-01-18:
@mikerocklewitz
Good evening Sir,
is there any chance of you providing information, preferably pictures, of where to install the sensor?

thanks in advance
kind regards

mikerocklewitz, 2022-01-18:
I apologize for never seeing this. I have a disassemble dead but should be modded board at home. Will get pictures tonight.

discip, 2022-01-18:
@Ralim
One more question though, which of the SOT23-5 do you recommend?
There is a bunch of 'em listed here https://files.pine64.org/doc/datasheet/pinecil/si7210-datasheet.pdf. 😅

Ralim, 2022-01-18:
You must match the one in the schematic `Si7210-B-00-IV(R)`

mikerocklewitz, 2022-01-19:
> I apologize for not seeing this. I have a disassemble dead but should be modded board at home. Will get pictures tonight.
> 
> Actually ill snap pics when i go on break on 3hrs. I wont keep yall waiting lol

Was that 3 hours? Im not gonna circle it, you can play where's wally or find the bodge work[image]

Ralim, 2022-01-19:
Ah you have the newer PCB revision.
Either way, it should be the only unpopulated spot the right size / shape :)

discip, 2022-01-25:
@mikerocklewitz
> Was that 3 hours? Im not gonna circle it, you can play where's wally or find the bodge work![20220118_182037.jpg]
> [image]

Sorry it is me one more time, but it seems that you've photographed 2 times the wrong side of the board 😅,  since the hall sensor belongs on the other side.
That photo made me think the sensor was preinstalled on my device. 🤣

Nevertheless thanks for trying!

Here is a photo @ritkit made showing the spot where the sensor belongs: [image]

mikerocklewitz, 2022-01-25:
> @mikerocklewitz
>
> Sorry it is me one more time, but it seems that you've photographed 2 times the wrong side of the board 😅,  since the hall sensor belongs on the other side.
> That photo made me think [...]

Maybe you should look closer. You can clearly see my hall sensor.

Perhaps your board is a different revision?

mikerocklewitz, 2022-01-25:
> Maybe you should look closer. You can clearly see my hall sensor.
>
> Perhaps your board is a different revision?

Im going to apologize 🤣🤣

I was looking at a chip i removed 
[image]

discip, 2022-01-25:
Ah that's a lot better!
## No need to apologize, every thing is fine be me! 😃👍
Just wanted to make this clear in case someone would see these photos thinking they are showing the populated hall sensor

mikerocklewitz, 2022-01-25:
If admin can clean this up please do

discip, 2022-01-25:
@mikerocklewitz
If you want, you could clean it up yourself by choosing **_edit_** after you click the _**3 dots**_ in the _**upper right corner**_ of the corresponding post. Than just swap the wrong photo with the one you posted last.😊

mikerocklewitz, 2022-01-26:
> @mikerocklewitz
> If you want, you could clean it up yourself by choosing **_edit_** after you click the _**3 dots**_ in the _**upper right corner**_ of the corresponding post. Then just swap the wrong photo with the one you posted last.😊

Yea. It may be a day or two til i get around to it though. I do not like github on my phone.

gregknight, 2023-02-08:
I guess I should’ve looked at the installation instructions before ordering the sensor, any tips on how to solder something to my soldering iron? 😂

mikerocklewitz, 2023-02-08:
> I guess I should’ve looked at the installation instructions before ordering the sensor, any tips on how to solder something to my soldering iron? 😂

2 Pinecils are better than one 💁‍♂️