Ralim/IronOS #1003: Pinecil stuck on start (after power is connected)

Firebie, 2021-06-25:
**Describe the bug**
Pinecil stuck on start (after power is connected) - oled looks like night with stars

**To Reproduce**
Connect pinecil to any power supply (pc usb (5v), pd (20v), dc(20)) - the iron stuck and oled looks like night with stars
No reaction on any button click

**Expected behavior**

**Details on your device:**
 - Device: Pinecil
 - Release:  2.15
 - Power adapter being used: PC USB, PD USB (20V), DC - 20V

Firebie, 2021-06-25:
If I comment these 2 lines, no stuck happens:
```c++
  /* definition and creation of POWTask - Power management for QC */
  osThreadStaticDef(POWTask, startPOWTask, osPriorityAboveNormal, 0, POWTaskStackSize, POWTaskBuffer, &POWTaskControlBlock);
  POWTaskHandle = osThreadCreate(osThread(POWTask), NULL);
```

Ralim, 2021-06-25:
This is abnormal, and should _not_ occur.
Can you please confirm what version you are running? (its not indicated in the initial question)
This would indicate that one of the calls in POWTask is handling the scheduler, which should ultimately cause a reset, but if its reproducible then it will not show anything.

Firebie, 2021-06-25:
Now I can't get to normal screen.
I took 'Pinecil_EN.hex'  from 2.15 release 'Pinecil.zip'.

Firebie, 2021-06-25:
2.15.40087E6

Ralim, 2021-06-25:
You absolutely _should not_ be having an issue like this on that release.
I would compile with PD turned off to check if you have killed your PD interface IC.

Firebie, 2021-06-25:
If I comment '#define POW_PD' - works without issues

Ralim, 2021-06-25:
Most likely you have a damaged FUSB302

Firebie, 2021-06-25:
Is it responsible only for PD or for both PD/QC?

Firebie, 2021-06-25:
Checked on 2.14 - no issues

Ralim, 2021-06-25:
its only responsible for PD.
On 2.14 does PD negotiate correctly at 20V?
I rewrote all of the i2c code from 2.14 to 2.15

Firebie, 2021-06-25:
On 2.14 I see 20V, I also see 20V at the latest version with commented '#define POW_PD'
I am not sure which protocol is used - charger is mdy-11-ez

Ralim, 2021-06-25:
That charger appears to use a USB-A connector, so has to be QC not PD.
Can you please test a PD charger. ❤️

Firebie, 2021-06-25:
Then I have no any PD charger available :-(

Ralim, 2021-06-25:
Hrmm, that will make testing a lot harder.
I'm guessing the FUSB302 is holding the I2C bus somehow.

Firebie, 2021-06-25:
Normal (without disable POW_PD) 2.15 build sometime starts ok (around 1 or 30).

mikerocklewitz, 2021-06-29:
Just throwing this out there....

I had to have a brand new pinecil that arrived 4 weeks ago replaced the day it arrived.  The defective one wouldnt read temperature and would get red hot. 

The new one came yesterday and I got brave with the bad one.  

What I discovered was poor soldering.  Cold joints somewhere between the heater end and the mcu.  When i would flex the pcb the resistance of the tip would change on the debug screen.  If i held the board just right it would start working correctly.

I ended up reflowing the whole board and adding leaded solder to everything from the mcu down. It works.

Not saying this is related to that, but its worth being aware if you are having a weird issue.

Firebie, 2021-07-25:
With USB-PD https://github.com/Ralim/IronOS/pull/1024 lib doesn't stuck

Ralim, 2021-07-25:
@Firebie 
What do you mean? 😂

Firebie, 2021-07-25:
Symptoms from the first post of this topic are gone with #1024

Ralim, 2021-07-25:
Hmm, odd 😓 
Though I did do a lot of cleaning up so errors are handled better. Surprising it fixed the issue though. (Happy that it does though).

Firebie, 2021-07-25:
Yes, I still need to check on PD power adapter, but it least it doesn't stuck on start up.

closed 2021-07-26 by Firebie